openhwfoundation/cva6 #1443: [TASK] Reformat RTL with VERIBLE

JeanRochCoulon, 2023-09-18:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

Use VERIBLE to reformat RTL

### Required Changes

Only CVA6 RTL is concerned

### Current Status

Todo

### Risks

On-going Github PR will be in conflict. A proposal can be to first reformat files which are not modified by on-going PRs.

### Prerequisites

none

### KPI (KEY Performance Indicators)

Number of files to be reformated

### Description of Done

All CVA6 RTL files are reformated and VERIBLE check from Github Action is pass.

ASintzoff, 2023-10-10:
- To mitigate Verible behaviour changes, use a unique version.
- Use of GitHub Action or GitLab pipeline (to be decided)
- If RTL code in a PR does not follow Verible rules, different options could exist.

1. PR rejected and the contributor has to format correctly the RTL
2. Add automatically a commit with Verible formatted files  and possibly used for the committed RTL in GitHub.

Option 1: every contributor needs to have Verible installation
Option 2: contributors will not be aware of formatting issues

jquevremont, 2023-10-10:
Can we let the user select the option?

Moschn, 2023-10-11:
Even though I am not participating that much in CVA6, I am providing my 2 cents on this: I would go for option 1. Currently, developers also have to see if CI passes and then make changes. Yes they can run that locally, but setting up all the checks that are run in the CI is orders of magnitudes more difficult instead of just installing a specific version of Verible and fixing lint issues. 

Besides, I have not yet seen an open source project employ option 2. It would also add unnecessary bloat of lint fixing commits in the history IMHO.

ASintzoff, 2023-10-11:
OK. Let's go for option 1. We will try to show in the CI results which RTL modifications are needed to be "compliant" to the expected format.

zarubaf, 2023-10-11:
> OK. Let's go for option 1. We will try to show in the CI results which RTL modifications are needed to be "compliant" to the expected format.

This https://github.com/chipsalliance/verible-linter-action could be interesting to you!

JeanRochCoulon, 2023-12-05:
@zarubaf  The task is done. But we setup a kind of process before closing definetly the task. A reviewer need to be identified to review the task and declare it as finished. Could you be this reviewer ?

JeanRochCoulon, 2023-12-07:
As discussed, the Verible command need to be added to the README.md to give possibility to users to execute it before submitting PR.

JeanRochCoulon, 2023-12-14:
The task is completed, I close it.

closed 2023-12-14 by JeanRochCoulon