openhwfoundation/cva6 #1447: [TASK] Extract PMP from MMU module

JeanRochCoulon, 2023-09-19:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

Today PMP dedicated to data is inside MMU module. Removing MMU removes PMP, which is not wanted. 

### Required Changes

The objective is to instantiate data PMP outside MMU.

### Current Status

TODO

### Risks

Be careful to well reconnect the exceptions information.

### Prerequisites

none

### KPI (KEY Performance Indicators)

none

### Description of Done

PR merged in cva6 master

JeanRochCoulon, 2024-02-08:
@mike, I tried to assign @CoralieAllioux to this task, but Bosch contributors are not included in the Github cva6 project. Could you include Coralie, Nicolas, Oliver and Jean-Christian ?

JeanRochCoulon, 2024-05-19:
@CoralieAllioux, the different MMUs have been unified last week. The PMP extraction is possible. Can you update us about your plan ?

CoralieAllioux, 2024-05-23:
Hi @JeanRochCoulon, it is still plan to contribute to the PMP extraction. Since the unified MMU is on the master branch, the design will be start (updated) ASAP. However, we miss to define some tests to provide with the RTL. We are working on it.
Does it answer your question?

JeanRochCoulon, 2024-05-23:
Good !

JeanRochCoulon, 2025-01-03:
Done by #2528

closed 2025-01-03 by JeanRochCoulon