openhwfoundation/cva6 #1458: cache miss

HimalSubedi787, 2023-09-19:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

https://planv.tech/2022/11/08/cva6s-instruction-and-writeback-data-cache/

I like this article that was written about CVA 6. 

Is there any way to perform an L1 cache miss in the CVA 6 system?

I am interested in measuring  L1 cache miss. Is there any miss status holding register (MSHR) in CVA 6?

suppamax, 2023-09-20:
You can easily write your own test, something like

```
// cachelines are 128bit long
#define uint128_t __uint128_t
volatile uint128_t data[16] __attribute__((section(".cache_share_region")));

void unrolled_read();

int main()
{
  long begin, end;

  begin = rdcycle();
  unrolled_read();
  end = rdcycle();

  return (end - begin)/(sizeof(data)/sizeof(data[0]));
}

void unrolled_read()
{
volatile uint64_t* dataptr = (uint64_t*)data;
*(dataptr+2*0);
*(dataptr+2*1);
*(dataptr+2*2);
*(dataptr+2*3);
*(dataptr+2*4);
*(dataptr+2*5);
*(dataptr+2*6);
*(dataptr+2*7);
*(dataptr+2*8);
*(dataptr+2*9);
*(dataptr+2*10);
*(dataptr+2*11);
*(dataptr+2*12);
*(dataptr+2*13);
*(dataptr+2*14);
*(dataptr+2*15);
}
```

Moschn, 2023-09-20:
CVA6 supports performance counters that allow the measurement of cache hits and misses: https://docs.openhwgroup.org/projects/cva6-user-manual/01_cva6_user/CSR_Performance_Counters.html

HimalSubedi787, 2023-09-20:
Thank you @Moschn Do you know how to access Event ID 1 and Event ID 2 of Event Selector? Where it has been implemented in the code?  I couldn't find it. Thank you.

Moschn, 2023-09-21:
The risc-v privilege specification describes how to access performance counters from software. Essentially you first select which counter you want to read by a write to an event selector CSR (`mhpmevent`). Then you can read the value by reading the respective `mhpmcounter`. For more information, take a look at the spec. A quick google search also returned examples such as https://stackoverflow.com/questions/60048337/accessing-hardware-performance-counters-in-risc-v.

If you want to see where it is implemented in the core itself then take a look at [core/perf_counters.sv](https://github.com/openhwgroup/cva6/blob/master/core/perf_counters.sv).

JeanRochCoulon, 2023-10-02:
Thanks @Moschn and @suppamax for your support. 
@HimalSubedi787 Are you satisfied by the answer ?   If it is the case, can you comment and close the issue ?

closed 2023-10-02 by HimalSubedi787