openhwfoundation/cva6 #1459: [TASK] Trap DV plan for embedded configuration

ASintzoff, 2023-09-20:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

Write trap DV plan for embedded configuration

### Required Changes

Add new DV plan

### Current Status

In Progress

### Description of Done

DV plan approved

ASintzoff, 2023-10-09:
Waiting Thales internal review

closed 2023-11-30 by JeanRochCoulon

JeanRochCoulon, 2023-11-30:
@AnouarZajni    Dvplan has been created by @ASintzoff    Can I ask you to review it ?   It is an opportunity to be aware of it before switching to its execution.   If you think it is completed, please comment and switch status from "done" to "finished".

AnouarZajni, 2023-12-06:
@JeanRochCoulon, review done