openhwfoundation/cva6 #1477: cpoy boot image stuck on fpga

hhhsiang, 2023-09-26:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I utilize the "make fpga" command to generate .mcs file and subsequently flash it onto the FPGA (Genesys2) using Vivado 2018.2. However, upon completion of the flashing process and subsequent reboot of the FPGA board, it becomes unresponsive, displaying the message "copying boot image done!" as shown in the accompanying images:
[image]
[image]

I also use `make fpga` at older version of cva6- ariane4.2, and it works successfully:
[image]

How could I fixed it? Thanks in advance!

JeanRochCoulon, 2023-10-02:
Hello @zchamski    I see you have pushed something to fix this PR. What is the status, does it fix the issue ?

zchamski, 2023-10-02:
This task should be assigned to @Gchauvon as I'm not involved with the FPGA (except for Spike build interactions).

cathales, 2023-10-02:
Hello @hhhsiang

Could you please tell us exactly which commit worked and which didn’t so that we can investigate?

This issue is maybe related to #1485

hhhsiang, 2023-10-02:
@Moschn ,
Yes, I followed the steps. I think the SD card should be ok because CVA6-Ariane 4.2 can correctly work on fpga, as following:
[image]
[image]
[image]
[image]

Moschn, 2023-10-02:
Hmm ok. Still very confusing to me that the GPT partitions print all zeros. It was not like this in the past. But maybe something changed in the bootloader in the mean time.

hhhsiang, 2023-10-02:
@Moschn, I'm uncertain about this. If the SD card hasn't been correctly set up, it shouldn't display the information I mentioned in my previous comment, right?

hhhsiang, 2023-10-02:
@cathales ,I did not try other commit. I just use `make fpga` command, and flash on to FPGA board. It can work correcly with CVA6-Ariane 4.2, but not with the latest version.