openhwfoundation/cva6 #1486: [BUG] No print out in QuestaSim simulation. Expected behavior or not?

0ena, 2023-09-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

I want to use Questa to run simulations on the CVA6 RTL.
To that end I setup the environment as suggested in your guidelines and I am using the following command to run the usual `Helllo CVA6` elf and obtain the final printout in the `transcript` file:

`make sim elf-bin=$RISCV/riscv64-unknown-elf/bin/pk target-options=hello.elf batch-mode=1 `

Alas, there is no print out as reported in #800. However, as mentioned [here](https://github.com/openhwgroup/cva6/issues/748#issuecomment-935927371_) this might be due to the updated (?) structure of the verification repos (core-v-verif, riscv-compliance, riscv-tests).

Consequently, as there is no print out, the TB finishes once it reaches the time out time.

I should mention that I have tried running this simulation with different versions of Questasim.

This is the transcript I get with QuestaSim 10.7:

```
 # ----------------------------------------------------------------
# UVM-1.1d
# (C) 2007-2013 Mentor Graphics Corporation
# (C) 2007-2013 Cadence Design Systems, Inc.
# (C) 2006-2013 Synopsys, Inc.
# (C) 2011-2013 Cypress Semiconductor Corp.
# ----------------------------------------------------------------
# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
#    Time: 0 ns  Iteration: 0  Instance: /ariane_tb/dut/i_ariane_peripherals/gen_uart/i_apb_uart/UART_RXFF
# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
#    Time: 0 ns  Iteration: 0  Instance: /ariane_tb/dut/i_ariane_peripherals/gen_uart/i_apb_uart/UART_TXFF
# [TRACER] Output filename is: trace_hart_0.log
# [UART]: bbl loader
#                                                                                                                                                                                                                                                      
# UVM_ERROR /mnt/nvme1T8/user1/newProj/csaw/cva6/corev_apu/tb/ariane_tb.sv(146) @ 40000110: reporter [Core Test] *** FAILED *** (tohost = 2147483647)
# ** Note: $finish    : /mnt/nvme1T8/user1/newProj/csaw/cva6/corev_apu/tb/ariane_tb.sv(151)
#    Time: 40000110 ns  Iteration: 3  Instance: /ariane_tb
# Saving coverage database on exit...
# End time: 21:46:55 on Sep 27,2023, Elapsed time: 0:09:23
# Errors: 0, Warnings: 2
```
and this is the error I get with QuestaSim 2021.2 and 2023.3:

```
** Error (suppressible): /mnt/nvme1T8/user1/newProj/csaw/cva6/core/include/cv64a6_imafdc_sv39_config_pkg.sv(11): (vlog-13389) Existing package 'cva6_config_pkg' would be overwritten.
/mnt/nvme1T8/user1/newProj/csaw/cva6/core/include/cv64a6_imafdc_sv39_config_pkg.sv(11): Verilog Compiler exiting
```

I have the following questions:

a) Is it expected to not have a print out anymore?

b) How do I know if the TB has finished successfully?

c) Is it expected for newer versions of Questa to error out, due to the updated "more parametrized" structure of CVA6?

Feel free to request more info if necessary.
Thank you in advance for your help and time.

Kind regards,
Nassos

JeanRochCoulon, 2023-10-02:
Hello @0ena 
The way to launch simulations has changed. If you follow the updated README.md we recommand to execute simulations by launching cva6.py script. If you find the same issue with comand, do not hesitate to report it. Regards

JeanRochCoulon, 2023-10-24:
Related to #1569

valentinThomazic, 2023-11-30:
@0ena, did you manage to solve your issue ? If you did, could you close it ?
This may help you use the proxy kernel otherwise : https://github.com/openhwgroup/cva6/issues/1358#issuecomment-1808134071
Please note that the support for the pk has been dropped and simulation should be ran using cva6.py

0ena, 2023-12-02:
Hi @valentinThomazic ,

not really, as I spent one or 2 days working on it but to no avail.
I also tried using the cva6.py but couldn't make it work.
This was back in September, since then my pipeline was full with deadlines, so I don't know the current state of the repo with respect to running simulations using Questa.
I will try to investigate a bit more at the end of December.

I must say that the guidelines with respect to running simulations (and the dependencies on repos) should be a bit more streamlined.
At least that's what I felt back in September.

Cheers,
Nassos

valentinThomazic, 2023-12-04:
Thank you for your feedback ! 
You should be able to run simulations on Questa with cva6.py by setting `export DV_SIMULATORS=questa-testharness` altough you will probably need to change the `verif/sim/Makefile` if you really want to use the pk.

mhamdy2712, 2023-12-24:
@0ena do you solve the problem i have the same here ?

mhamdy2712, 2023-12-25:
I solved it by adding -suppress 13389 to questa-flags in make file

JeanRochCoulon, 2023-12-27:
Good ! Did you use cva6.py to run your simulation ?   If it is the case, it would be nice to submit a PR to solve the issue.

closed 2024-01-27 by JeanRochCoulon