openhwfoundation/cva6 #1489: make verilate failed

hhhsiang, 2023-09-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I've come across a problem much like the one described in https://github.com/openhwgroup/cva6/issues/1410. When I run `make verilate`, it displays the exact same error.
[image]

I've been struggling with this issue for a while. Any help in finding a solution would be very welcome. Thank you in advance!

JeanRochCoulon, 2023-10-02:
Hello @hhhsiang 
Which Verilator version do you use ?
Regards

hhhsiang, 2023-10-02:
Hi,@JeanRochCoulon
I use verilator version 4.110

JeanRochCoulon, 2023-10-02:
Please refer to the README.md CVA6 repository file, the supported Verilator is not more 4.110. If it works with the recommanded version, do not hesitate to close the issue.

hhhsiang, 2023-10-02:
@JeanRochCoulon  I'm sorry, but I couldn't find the information about the required Verilator version in the [README.md](https://github.com/openhwgroup/cva6) file.

JeanRochCoulon, 2023-10-02:
You are right. The supported Verilator version is 5.008. README.md need to be updated.

JeanRochCoulon, 2023-10-02:
I have created #1492 to add information about Verilator version

hhhsiang, 2023-10-02:
@JeanRochCoulon, thanks for you help!
However, I'm not sure how to update the version of verilator. Should I use `./ci/install-verilator.sh` , and change the version to 5.008? Like this:
[image]

JeanRochCoulon, 2023-10-02:
You have to execute the verif/regress/install-verilator.sh file. The Verilator5.008 will be installed.

hhhsiang, 2023-10-02:
@JeanRochCoulon, I have tried it, but it show this:
[image]

JeanRochCoulon, 2023-10-02:
The README.md recommands to execute verif/regress/install-cva6.sh.
I suggest to execute it.

hhhsiang, 2023-10-02:
@JeanRochCoulon, it appears to have been installed successfully, but when I run the command `verilator -version`, it still displays version 4.110, as follows:
[image]

Moschn, 2023-10-02:
Your linux/PATH is not configured properly. There are many ways to fix this, but they are not really related to CVA6. I would suggest to get familiar with the PATH environment variable and how your shell searches for executables. A quick google search results in simple tutorials: 

- https://www.digitalocean.com/community/tutorials/how-to-view-and-update-the-linux-path-environment-variable
- https://www.linfo.org/path_env_var.html
- https://astrobiomike.github.io/unix/modifying_your_path

hhhsiang, 2023-10-04:
@Moschn, I can now execute `bash verif/regress/smoke-tests.sh` and `bash verif/regress/dv-riscv-tests.sh` without encountering any errors. However, I continue to encounter the same error when executing `make verilate`. Is it conceivable that there may be an issue with the `make verilate` command?

Moschn, 2023-10-04:
I just successfully ran `make verilate` on verilator 4.110 (not the suggested version but still works). I do not know where your problem lies. Maybe you have a very old gcc version. I have never encountered this issue.

hhhsiang, 2023-10-04:
@Moschn ,did you install verilator from `./ci/install-verilator.sh`?

Moschn, 2023-10-04:
No i installed manually using the official instructions from verilator

hhhsiang, 2023-10-06:
@JeanRochCoulon, what if I want to run user-space applications? How should I proceed? According to the CVA6 README, should I execute `make verialte` followed by running `work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf`?
[image]

Thanks for your help!

JeanRochCoulon, 2023-10-06:
Please create another GihHub issue when the topic is different.

Please refer to #1505. Make verilate is a deprecated command. The supported command is cva6.py. Please close the issue if you succeed in simulating with cva6.py.

JeanRochCoulon, 2023-10-06:
BTW #1499 is related to your new topic.

hhhsiang, 2023-10-06:
https://github.com/openhwgroup/cva6/pull/1505#issuecomment-1750022248

valentinThomazic, 2023-12-05:
Hey @hhhsiang did you manage to close your issue ? 
If that is the case, could you close it ?

closed 2023-12-06 by hhhsiang