openhwfoundation/cva6 #1490: [BUG] <title>

MaistoV, 2023-09-29:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I noticed that the PMP exception reporting in CVA6 returns the physical address as exception `tval`, e.g., [here](https://github.com/openhwgroup/cva6/blob/master/core/mmu_sv39/mmu.sv#L369C25-L369C41) and [here](https://github.com/openhwgroup/cva6/blob/master/core/mmu_sv39/mmu.sv#L406), where `lsu_paddr_o` is the address the PMP checks are performed against, i.e., [here](https://github.com/openhwgroup/cva6/blob/master/core/mmu_sv39/mmu.sv#L436), so it must be the final translated physical address.

This is explicitly against the privileged spec: 

> 3.1.16 Machine Trap Value Register (mtval):
> If mtval is written with a nonzero value when a breakpoint, address-misaligned, access-fault, or
page-fault exception occurs on an instruction fetch, load, or store, then mtval will contain the
faulting virtual address.
> When page-based virtual memory is enabled, mtval is written with the faulting virtual address,
even for physical-memory access-fault exceptions.
>  
>  4.1.9 Supervisor Trap Value (stval) Register:
> If stval is written with a nonzero value when a breakpoint, address-misaligned, access-fault, or
page-fault exception occurs on an instruction fetch, load, or store, then stval will contain the
faulting virtual address.

Is there a reason behind this choice or is it just a latent bug?

JeanRochCoulon, 2023-10-02:
Thank you @MaistoV for this bug report.  I asign it to @ASintzoff who will start TRAP veriifcation.

Moschn, 2023-10-03:
Seems like a bug to me. Feel free to propose a fix. 

However, your second link does not point to the correct line. I believe you mean lines 369 and 379. Your second link points to 406, where a virtual address is loaded into `lsu_exception_o`.

MaistoV, 2023-10-03:
> Seems like a bug to me. Feel free to propose a fix.

Just replacing `lsu_paddr_o` with `lsu_vaddr_q` would work. Should I open a PR?

> However, your second link does not point to the correct line. I believe you mean lines 369 and 379. Your second link points to 406, where a virtual address is loaded into `lsu_exception_o`.

Thank you for pointing out the the mistaken line in the second link. It is fixed now.

JeanRochCoulon, 2023-10-03:
Hello,
Yes a PR need to be submitted. In that way the CI will check it.
Thanks to both of you

JeanRochCoulon, 2023-10-06:
If the commit solves the issue, thank you to close this GitHub issue.

closed 2023-10-06 by MaistoV