openhwfoundation/cva6 #1491: OpenSBI and U-BOOT UART doesn't print everything correctly

jimaandro, 2023-10-01:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I have managed to boot via GDB, it boots OpenSBI and then goes to u-boot. But the problem is that it doesn't print everything (e.g. it misses characters) like this:
[image]

I also noticed that the [driver of OpenSBI](https://github.com/riscv-software-src/opensbi/blob/be245acfffa297b5ed4e0c7bb473a6bd55231bf8/lib/utils/serial/uart8250.c) is a bit different from CVA6 [uart drivers](https://github.com/openhwgroup/cva6/blob/master/corev_apu/fpga/src/bootrom/src/uart.c) and also it doesn't match with the [uart hardware](https://github.com/pulp-platform/apb_uart/blob/ac3461ce23832e02903b9d2d7d4edd7fcb89bd92/src/apb_uart.vhd) (UART 16750)
So what should I do to print correctly to uart?

JeanRochCoulon, 2023-10-02:
@sjthales  I forward you this issue in the case you can support @jimaandro. Cheers

jimaandro, 2023-10-02:
Actually it got fixed. I really don't know how. It just works now

closed 2023-10-02 by jimaandro