openhwfoundation/cva6 #1499: [BUG] install riscvpk failed

RonxBulld, 2023-10-04:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

in the description of [#1358](https://github.com/openhwgroup/cva6/issues/1358), riscv64-unknown-elf to be replaced by riscv-non-elf, but my build system always talk:
```
checking build system type... x86_64-pc-linux-gnu
checking host system type... Invalid configuration `riscv-none-elf': machine `riscv-none' not recognized
configure: error: /bin/bash ../scripts/config.sub riscv-none-elf failed
```
And the $PATH is:
```
/home/cva/tools/bin:/home/cva/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
```

JeanRochCoulon, 2023-10-06:
@sjthales  Are you familiar with riscvpk command ?   I twould be nice to help @RonxBulld and @hhhsiang if it is the case.

JeanRochCoulon, 2023-10-24:
Related tash #1569

valentinThomazic, 2023-11-30:
Hey @RonxBulld, did you manage to solve your issue ? If you did, could you close it ?

JeanRochCoulon, 2023-12-05:
Hi @RonxBulld    Thanks @valentinThomazic having given recommendation to help. But as proxy kernel is not maintained by CVA6 repository, that's why I propose to close the Github issue and I invite you to contribute to add this feature.

closed 2023-12-05 by JeanRochCoulon