openhwfoundation/cva6 #1500: Error when run "make verilate"

Chentai-Y, 2023-10-04:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi Assignees,

I met the error when run make verilate:

```
Makefile:80: "Specified QuestaSim version (questa-2021.2) not found in PATH /home/sy/intelFPGA_lite/22.1std/questa_fse/bin:/home/sy/intelFPGA_lite/22.1std/questa_fse/linux_x86_64/mgls/bin:/home/sy/riscv/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin"
mkdir -p build
sh: 276: [: Illegal number: 
bender-init: Warning: No release for platform 'x86_64-linux-gnu-ubuntu22.04' version '0.23.1' found, using latest.
Successfully installed bender 0.27.3 in '/home/sy/source-code/ara/hardware'.
bender 0.27.3 available.
rm -rf build/verilator; mkdir -p build/verilator
./bender script verilator -t rtl -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define RVV_ARIANE=1 > build/verilator/bender_script_default
warning: Name issue with "cva6", `export_include_dirs` not handled
	Could relate to name mismatch, see `bender update`
/home/sy/source-code/ara/install/verilator/bin/verilator -f build/verilator/bender_script_default           \
  -GNrLanes=4                                                         \
  -O3                                                                           \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LATCH                                                                    \
  -Wno-LITENDIAN                                                                \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  -Wno-ENUMVALUE                                                                \
  -Wno-COMBDLY \
  --hierarchical                                                                \
  tb/verilator/waiver.vlt                                                       \
  --Mdir build/verilator                                                       \
  -Itb/dpi                                                                      \
  --compiler clang                                                              \
  -CFLAGS "-DTOPLEVEL_NAME=ara_tb_verilator"                                        \
  -CFLAGS "-DNR_LANES=4"                                              \
  -CFLAGS -I/home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  ""                                                             \
  -LDFLAGS "-lelf"                                                              \
  ""                                                              \
  --exe                                                                         \
  /home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/sy/source-code/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
                                       \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
make[1]: Entering directory '/home/sy/source-code/ara/hardware/build/verilator'
make[2]: Entering directory '/home/sy/source-code/ara/hardware'
/home/sy/source-code/ara/install/verilator/bin/verilator -f build/verilator/Vlane_e_hierMkArgs.f
sh: 1: exec: /home/sy/source-code/ara/install/verilator/share/verilator/verilator_bin: not found
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/sy/source-code/ara/install/verilator/share/verilator/verilator_bin -f build/verilator/Vlane_e_hierMkArgs.f
make[2]: *** [build/verilator/Vara_tb_verilator_hier.mk:361: hier_launch_verilator] Error 127
make[2]: Leaving directory '/home/sy/source-code/ara/hardware'
make[1]: *** [Vara_tb_verilator_hier.mk:370: Vlane_e/lane_e.sv] Error 2
make[1]: Leaving directory '/home/sy/source-code/ara/hardware/build/verilator'
%Error: make -C build/verilator -f Vara_tb_verilator_hier.mk  -j 1  hier_verilation exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/sy/source-code/ara/install/verilator/bin/verilator_bin -f build/verilator/bender_script_default -GNrLanes=4 -O3 -Wno-BLKANDNBLK -Wno-CASEINCOMPLETE -Wno-CMPCONST -Wno-LATCH -Wno-LITENDIAN -Wno-UNOPTFLAT -Wno-UNPACKED -Wno-UNSIGNED -Wno-WIDTH -Wno-WIDTHCONCAT -Wno-ENUMVALUE -Wno-COMBDLY --hierarchical tb/verilator/waiver.vlt --Mdir build/verilator -Itb/dpi --compiler clang -CFLAGS -DTOPLEVEL_NAME=ara_tb_verilator -CFLAGS -DNR_LANES=4 -CFLAGS -I/home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -CFLAGS -I/home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -CFLAGS -I/home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -LDFLAGS -lelf  --exe /home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/dpi_memutil.cc /home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/sv_scoped.cc /home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/verilator_memutil.cc /home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.cc /home/sy/source-code/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc /home/sy/source-code/ara/hardware/tb/verilator/ara_tb.cpp --cc --top-module ara_tb_verilator
make: *** [Makefile:168: build/verilator/Vara_tb_verilator] Error 2

```

My verilator version is 5.012, and I also tried to expand the memory for Ubuntu 22.04.... They didn't work.

Could you please take a look at the problem? Thanks!

JeanRochCoulon, 2023-10-06:
Please refer to #1505. Make verilate is a deprecated command. The supported command is cva6.py. Please close the issue if you succeed in simulating with cva6.py.

valentinThomazic, 2023-12-01:
Hi, did you manage to solve your issue ?

JeanRochCoulon, 2023-12-05:
@Chentai-Y   As the "make verilate" command is deprecated and the discussion is down for several weeks, I propose to close the issue.