openhwfoundation/cva6 #1502: [BUG] The div instruction creates a time side channel in CVA6 when the divisor is zero.

jooohndoe, 2023-10-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

### Summary
Hi all, we have found a timing vulnerability in CVA6 caused by div instruction when the divisor is zero. A similar vulnerability, termed Zero Dividend Injection (ZDI), has been reported for X86 Intel processors, as mentioned on the Microsoft Revizor page [https://microsoft.github.io/sca-fuzzer/](https://microsoft.github.io/sca-fuzzer/).

### Details
We use chipyard 1.7.1 as the simulation platform with the commit 705c48a of CVA6. We use VCS for simulation.
When the divisor is zero, the execution time taken is greater than when it is not equal to zero. For example, consider the following cases:

- Test 1 takes 92 ns for simulation
```
       li     a4, 0
       li     a7, 1333
       divuw  t3, a7, a4
```

- Test 2 takes 36 ns for simulation
```
       li     a4, 3
       li     a7, 1333
       divuw  t3, a7, a4
```

### PoC
Sequences of instructions can trigger the vulnerabilities that have been shown above.

### Impact
This dependence is capable of causing a Spectre type vulnerability by using the following code snippet:
```
divuw a4, key, attacker's_no;
divuw t3, a7, a4;
```
Here, two cases arise:

- Case 1:

`key < attacker's_value`
This places `a4 = 0`, thereby causing the code snippet to take greater execution time to complete. Hence, the attacker can tell that the key value is less than their own provided value.

- Case 2:

`key >= attacker's value`
This places `a4 != 0`, thereby taking less execution time. Hence, the attacker can tell that the key value is less than their own provided value. By binary search, the attacker can then arrive at the correct key value. This impacts any security-sensitive algorithm using a `divuw` instruction like cryptographic implementations, etc.

Moschn, 2023-10-06:
Hi,

Thanks for your report. First a quick disclaimer: CVA6 does not promise to be timing channel free. There are various timing channels in the design currently such as the cache and operand-dependent arithmetic instructions such as division.

I do not quite understand your 'spectre' type vulnerability. If a secret value is in a register, and the attacker controls the binary, then why not just read the secret value and print it? Do you mean a snippet that first speculatively loads the secret value from memory? 

```
lw a2, address_to_protected_key
divuw a4, a2, attacker's_no;
divuw t3, a7, a4;
```

AFAIK, this is not an issue in CVA6 because the core does not speculate past memory loads.

jooohndoe, 2023-10-06:
Thank you for your reply,
Indeed, the code snippet identified as a potential exploit can be executed within any transient window. The attacker's approach involves loading the key transiently and subsequently performing div instructions on the loaded key. The crucial factor here is that if the key is smaller than the value selected by the attacker, the execution of these instruction sequences will take longer compared to when the key is larger than the attacker's chosen value. This instruction can then be followed by an instruction which transient effect would be visible to attacker, such as a load instruction to a dummy address. If the execution time taken of div is less, this secondary instruction would execute within the transient window.
This enables the attacker to encode information within a time channel, ultimately leading to the leakage of critical information.

Please let us know if this is clear.
Thanks

zarubaf, 2023-10-07:
Thanks a lot for the detailed report. I am not entirely sure I get it though, but:

1. I second what @Moschn said that CVA6 does not make any promises to be side-channel free.
2. Shouldn't any sensitive (i.e., algorithms using keys) **not** use `div`/`rem` instructions? If you look at the [RISC-V `Zkt`](https://github.com/rvkrypto/riscv-zkt-list/blob/main/zkt-list.adoc#rvm-multiply) extension `div`/`rem` can have data-dependent execution time. My understanding was that any crypto algorithm dealing with such things should use data-independent instructions. I don't wanna make any promises, but from a first-glance CVA6 actually implements `Zkte` (not verified, and by pure simplicity of the pipeline).

How would a fix to such an attack look? Thanks again for the work and the detailed report!

JeanRochCoulon, 2023-10-17:
This enhancement has already been listed in the requirement specification https://docs.openhwgroup.org/projects/cva6-user-manual/02_cva6_requirements/cva6_requirements_specification.html as ISA-110 requirement.

jooohndoe, 2023-10-19:
Hi all, 

Thank you for your feedbacks and detailed explanations! 

**Division by zero** without exception will create a serious security whole. We are contacting RISC-V organization to clarify why they don't want to throw such exceptions. From the RISCV ISA document, they explain that they want to simplify the implementation, which is vague for us.

But still appreciate providing such an elegant open-sourced processor design with many advanced features and actively maintain the design.