openhwfoundation/cva6 #1511: Linux Kernel stops initialization at random: crng init done

jimaandro, 2023-10-07:
Hi
I have a cva6 design, I am not using SPI or ethernet. the OpenSBI + u-boot are working and now I am trying to port the Linux Kernel.  I use OpenOCD + GDB to boot. The DTS file is this:

```
 /dts-v1/;

/ {
	#address-cells = <0x02>;
	#size-cells = <0x02>;
	compatible = "eth,ariane-bare-dev";
	model = "eth,ariane-bare";

	chosen {
			stdout-path = "/soc/uart@10000000:115200";
	};

	cpus {
		#address-cells = <0x01>;
		#size-cells = <0x00>;
		timebase-frequency = <0x17d7840>;

		cpu@0 {
			clock-frequency = <0x2faf080>;
			device_type = "cpu";
			reg = <0x00>;
			status = "okay";
			compatible = "eth, ariane", "riscv";
			riscv,isa = "rv64imafdc";
			mmu-type = "riscv,sv39";
			tlb-split;

			interrupt-controller {
				#interrupt-cells = <0x01>;
				interrupt-controller;
				compatible = "riscv,cpu-intc";
				phandle = <0x01>;
			};
		};
	};

	memory@80000000 {
		device_type = "memory";
		reg = <0x0000 0x80000000 0x00 0x40000000>;
	};

	soc {
		#address-cells = <0x02>;
		#size-cells = <0x02>;
		compatible = "eth,ariane-bare-soc", "simple-bus";
		ranges;

		clint@2000000 {
			compatible = "riscv,clint0";
			interrupts-extended = <0x01 0x03 0x01 0x07>;
			reg = <0x00 0x2000000 0x0 0x10000>;
			reg-names = "control";
		};

		interrupt-controller@c000000 {
			compatible = "riscv,plic0";
			interrupt-controller;
			#interrupt-cells = <0x01>;
			interrupts-extended = <0x01 0x0b 0x01 0x09>;
			reg = <0x00 0xc000000 0x00 0x40000000>;
			riscv,max-priority = <0x07>;
			riscv,ndev = <0x01>;
			phandle = <0x02>;
		};

		uart@10000000 {
[... 208 lines cut ...]
[   42.556954] usbcore: registered new interface driver usbhid
[   42.575851] usbhid: USB HID core driver
[   42.575851] usbhid: USB HID core driver
[   42.735824] NET: Registered protocol family 10
[   42.735824] NET: Registered protocol family 10
[   42.835522] Segment Routing with IPv6
[   42.835522] Segment Routing with IPv6
[   42.853287] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   42.853287] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   42.957518] NET: Registered protocol family 17
[   42.957518] NET: Registered protocol family 17
[   43.002516] Key type dns_resolver registered
[   43.002516] Key type dns_resolver registered
[   43.166242] Freeing unused kernel memory: 184K
[   43.166242] Freeing unused kernel memory: 184K
[   43.189748] Run /init as init process
[   43.189748] Run /init as init process
[   63.925773] random: dd: uninitialized urandom read (512 bytes read)
[   63.925773] random: dd: uninitialized urandom read (512 bytes read)
[   77.198244] random: crng init done
[   77.198244] random: crng init done

```

Why is this happening?

jimaandro, 2023-11-18:
The problem was my modified PLIC.

closed 2023-11-18 by jimaandro

maxbjurling, 2023-11-20:
@jimaandro would you like to elaborate on what the problem was with the PLIC?

jimaandro, 2023-11-20:
> @jimaandro would you like to elaborate on what the problem was with the PLIC?

Sure!
I had modified the number of sources in PLIC (I changed it to 3). After that, even though OpenSBI and U-Boot were working fine, the kernel couldn't initialize the IRQs, so it got stuck. The reason why I changed the PLIC NumSources is that I had modified the whole project and had to solve some other problems that were created.

jason23g, 2024-12-16:
hello @jimaandro,

Could you post the fixed dts containing the right plic informations.

Thanks in advance for your time.