openhwfoundation/cva6 #1514: [BUG Verif env] get_seed() in cva6.py missing declaration

CoralieAllioux, 2023-10-09:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

While launching `sh verif/regress/smoke-gen_tests.sh`, it seems that the function _get_seed()_ called in _do_simulate()_ (line 251 of _cva6.py_) is not declared anywhere. 
I fixed it locally by creating my own _get_seed()_ function. 
Could it be possible that a file declaring this function or an import is missing?

Trace:
```Traceback (most recent call last):
  File "<path>/cva6/verif/sim/cva6.py", line 1225, in <module>
    main()
  File "<path>/cva6/verif/sim/cva6.py", line 1199, in main
    gen(matched_list, cfg, output_dir, cwd)
  File "<path>/cva6/verif/sim/cva6.py", line 321, in gen
    do_simulate(sim_cmd, test_list, cwd, argv.sim_opts, argv.seed_yaml, argv.seed, argv.csr_yaml,
  File "<path>/cva6/verif/sim/cva6.py", line 251, in do_simulate
    rand_seed = get_seed(seed)
NameError: name 'get_seed' is not defined
```

JeanRochCoulon, 2023-10-10:
Good @CoralieAllioux to see you in GitHub !
@AyoubJalali, as father of smoke-gen_tests.sh, can you help ?

AyoubJalali, 2023-10-10:
> @AyoubJalali, as father of smoke-gen_tests.sh, can you help ?

this related to update the RISCV-DV, there's a MR for this

JeanRochCoulon, 2023-10-10:
Ayoub,
Do you think Coralie is using the uptodate version of riscv-dv (what is in your MR) while the right version to be used is older. This means, Coralie, that your riscv installation is not good.

AyoubJalali, 2023-10-10:
> Ayoub, Do you think Coralie is using the uptodate version of riscv-dv (what is in your MR) while the right version to be used is older. This means, Coralie, that your riscv installation is not good.

to work with a working version, use this HASH :  0ce85d3

CoralieAllioux, 2023-10-10:
Then, it would be useful to update file _cva6/verif/regress/install-riscv-dv.sh_ with correct HASH.
I'll try with HASH 0ce85d3

AyoubJalali, 2023-10-10:
> Then, it would be useful to update file _cva6/verif/regress/install-riscv-dv.sh_ with correct HASH.

Yes, but we're working on updating the HASH to work with latest version of RISCV-DV

CoralieAllioux, 2023-10-10:
Thank you @AyoubJalali , I confirm it currently works on my side with HASH 0ce85d3 for riscv-dv.
Do you have an estimation when the MR you're talking about will be effective?

AyoubJalali, 2023-10-10:
> Thank you @AyoubJalali , I confirm it currently works on my side with HASH 0ce85d3 for riscv-dv. Do you have an estimation when the MR you're talking about will be effective?

If there's any problem it will be this week

JeanRochCoulon, 2023-10-10:
@cathales

cathales, 2023-10-10:
To me the cause of the issue is that the "official" supported hash in [`cva6/verif/regress/install-riscv-dv.sh:26`](https://github.com/openhwgroup/cva6/blob/master/verif/regress/install-riscv-dv.sh#L26) is not the one used in the CI infrastructure. It is overwritten by an environment variable (might be a workaround for you @CoralieAllioux). This variable points to a version locally available in our servers. It is three year old, with a few patches from us.

It would be better if the same "source" was used to define the DV hash used by everyone and the CI, and that this version is not too old. So we are transforming DV into a submodule with an up-to-date hash. Then Dependabot will be able to keep it updated. 

Current status on this work:

- @AyoubJalali is bringing fixes relative to this update (we had the same errors as @CoralieAllioux when we only updated the hash to the most recent version)
- I have opened a PR https://github.com/chipsalliance/riscv-dv/pull/960 17 hours ago so that our fixes are integrated upstream

JeanRochCoulon, 2023-10-24:
Related to #1569

AyoubJalali, 2023-11-14:
Fixed by PR #1576

closed 2023-11-14 by AyoubJalali