openhwfoundation/cva6 #1515: [BUG Verif env] riscv_core_setting.sv for rv32imac target not up to date

CoralieAllioux, 2023-10-09:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

While launching `sh verif/regress/smoke-gen_tests.sh` on default target `cv32a60x`, it uses file `cva6/verif/env/corev-dv/target/rv32imac/riscv_core_setting.sv`. It seems that file is not up to date anymore with current riscv-dv version.

I fixed it locally by adding missing parameter, to be aligned with others riscv_core_setting.sv in riscv-dv:
```
// GPR setting
parameter int NUM_FLOAT_GPR = 32;
parameter int NUM_GPR = 32;
parameter int NUM_VEC_GPR = 32;

// Minimum size of a sub - element, which must be at most 8 - bits.
parameter int SELEN = 8;

// Maximum size of a single vector element(encoded in vsew format)
parameter int VELEN = int'($ln(ELEN) /$ln(2)) - 3;

// Maxium LMUL supported by the core
parameter int MAX_LMUL = 8;

// Implementation - specific custom CSRs
bit[11:0] custom_csr[] = {
};
```

AyoubJalali, 2023-10-10:
we're changing the target, because the embedded version does not have the Atomic extension

CoralieAllioux, 2023-10-10:
If target _cv32a60x_ is deprecated, it would be nice to change default values in scripts in _cva6/regress/_. Since we're still exploring the verification environment, it's not so obvious for new comers what should work or not.

Anyway, for people who actually use _rv32imac_, it could be useful to update _riscv_core_settings.sv_ accordingly. What do you think?

Thank you for your answer, we'll focus more on target _cv32a6_embedded_, which seems to be the most used and up to date.

AyoubJalali, 2023-10-10:
> Anyway, for people who actually use _rv32imac_, it could be useful to update _riscv_core_settings.sv_ accordingly. What do you think?

make sense

JeanRochCoulon, 2023-10-24:
@CoralieAllioux  Submit a (first) PR !!  ;-)

JeanRochCoulon, 2023-10-24:
Realted to #1569

CoralieAllioux, 2023-10-24:
> @CoralieAllioux Submit a (first) PR !! ;-)

I'll be my pleasure ;)

valentinThomazic, 2023-12-07:
Hey @CoralieAllioux, I was wondering if you were still planning on doing your PR and what was the statue of the issue.

CoralieAllioux, 2023-12-11:
Hi @valentinThomazic, I'm still planning to do the PR. Unfortunately, I'm still waiting the green light from my company to create PR. I'll do it as soon as possible but I cannot tell when it'll be.

closed 2024-02-17 by github-actions[bot]