openhwfoundation/cva6 #1525: [Improvement] [Verif env] tohost_addr more generic for vcs_uvm_run in cva6/verif/sim/Makefile

CoralieAllioux, 2023-10-10:
While using the rule _vcs_uvm_run_ from `cva6/verif/sim/Makefile`, I noticed that the command which determines **tohost_addr** is not generic enough if the prefix differs: `$(shell $$RISCV/bin/riscv32-unknown-elf-nm -B $(elf) | grep -w tohost | cut -d' ' -f1)`

I would suggest to use an environment variable instead. For example, `Common.mk` from `cva6/verif/core-v-verif/mk` sets **$RISCV_PREFIX**.
We could then imagine: 
```
+tohost_addr=$(shell $(RISCV_PREFIX)nm -B $(elf) | grep -w tohost | cut -d' ' -f1) \
```

It is already the case for rules **vcs-testharness** and **veri-testharness** of the same file: `+tohost_addr=$(shell $$RISCV/bin/${CV_SW_PREFIX}nm -B $(elf) | grep -w tohost | cut -d' ' -f1)`

JeanRochCoulon, 2023-10-17:
@zchamski Can you give your opinion on this ?!?

zchamski, 2023-10-17:
Clearly, this is an oversight.  The fix is straightforward and can be implemented rapidly using the `CV_SW_PREFIX` variable (to be consistent with the other sim targets).

JeanRochCoulon, 2023-10-24:
Related to #1569

zchamski, 2023-11-24:
Fixed, merged into master in commit 220f534b6.