openhwfoundation/cva6 #1533: [BUG] UVM failed on questa Sim

AnikBalo, 2023-10-16:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi, I was trying to simulate CVA6 in questaSim. 
While running the command:
"make sim elf-bin=<elf_path_of_dhrystone>"
 I am not getting any error but UVM FAILED. Actually, I want to generate the activity file from the simulation. Any idea if this is okay or how to debug this issue? Thanks!  

The output: 
[image]

JeanRochCoulon, 2023-10-17:
Hello
As you can read in README.md, make sim is not the right way to simulate the CVA6. You should use cva6.py command. But on my side, I use only VCS and verilator, I cannot confirm the Questasim is fully functional, but I wold be pleased to get your feedback.

AnikBalo, 2023-10-18:
@JeanRochCoulon I thnik CVA6 section has been removed from [core-v-verif](https://github.com/openhwgroup/core-v-verif#core-v-verif) repository.

JeanRochCoulon, 2023-10-18:
The README.md with the commands to be executed is in cva6 repo (and not core-v-verif)

valentinThomazic, 2023-12-01:
Hi, did you manage to solve your issue ?

JeanRochCoulon, 2023-12-05:
Hello @AnikBalo    Today Questa is not supported by CVA6 repository. If you are ready to contribute the target questa-testharness can be updated to work. I propose to close this issue becasue it is not supported.

closed 2023-12-05 by JeanRochCoulon