openhwfoundation/cva6 #1541: [TASK] New Release

Moschn, 2023-10-18:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

The repository lists an old release from the Ariane days (Ariane v4.2). Since many people rely on the release, especially for the prebuilt FPGA image, it would be nice to create a new release with all the fixes and improvements of the past 4 (!) years. Even if this new release is just an alpha or beta version, it is still beneficial to create such a release. People wanting to test certain RISC-V software on RISC-V do not necessarily want to set up an entire Xilinx Viviado toolchain + all the other requirements for CVA6. A simple download of an FPGA image is significantly easier. 

On another note, it would be nice if the gitlab CI would make the artifacts that it produced for the builds available. Then, we can have a "nightly" FPGA image. The images are built and tested already, so making them available for download should not add any processing overhead.

### Required Changes

The main things that would need to be done for such a release are:

- Decide on the configurations to be a part of the release
  - I assume most people are interested in `cv32a6_embedded` and `cv64a6_imafdc_sv39` 
- Create all the build artifacts (this is partially already done in the GitLab CI)
- Test the build artifacts (verify Linux boot on 32 and 64bit versions)

### Current Status

Not started

### Description of Done

A new release on the Github repository

JeanRochCoulon, 2023-10-19:
@Moschn   Are you available to discuss the point during a CVA6 meeting ?   Before accepting the task and add it to the Kanban board, I would be pleased to grant it with the cva6 project stakeholders. Meeting is on Tuesday, did you receive an invitation ?

Gchauvon, 2023-10-30:
Having a new release on a "recent" commit is a **must have**. Once the commit is selected I can easily generate the artifacts and the bitstream and put it on github. I have no opinion on which is a better commit.

Today the Linux boot is only done in 32bits in CI on the `cv32a60x` configuration since the `cv32a6_embedded` cannot boot linux (no MMU). Testing linux boot for 32 and 64 bits is possible with only one FPGA but it requires the improvement of the scripts already existing to push the linux images via JTAG rather than via the SD card. So it will differ from the common flow described in the README.

JeanRochCoulon, 2024-05-19:
@ASintzoff   This task maybe be ready to be closed since a new release has been posted.  What do you think ?

closed 2024-05-20 by ASintzoff