openhwfoundation/cva6 #1547: [BUG] Multiple compressed instructions and the MV instruction create time side-channels in CVA6

jooohndoe, 2023-10-19:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

### Summary
Hi all, we have found multiple compressed instructions that violate the [_RISC-V Zkt "Constant Time" Instruction List_](https://github.com/rvkrypto/riscv-zkt-list/blob/main/zkt-list.adoc). These instructions fail to prevent leakage of sensitive parameters via the direct timing channel and hence should be considered serious security vulnerabilities. These instructions are: `c.mv, c.add, c.and, c.or, c.xor, c.sub, c.addw, c.subw`. Also, the instruction `mv` causes a time side-channel. Though the mv instruction is not on the list, but we suspect that those vulnerabilities share the same root cause. 

### Details
When the `rs2` register's value is zero, the execution time of these instructions is less than when the `rs2 ` register's value is not equal to zero. We use two registers (`a3, a4`) among eight popular registers, mentioned by RISCV ISA, to create tests. We use [chipyard 1.7.1](https://chipyard.readthedocs.io/en/stable/) as the simulation platform with the commit 705c48a of CVA6 and VCS for simulation. We used the same tests for Rocket Core and Boom and only observed execution time difference from CVA6.

Below are two test examples:

- test1, set the value of `rs2` to be zero, then test any instructions mentioned above
we observe 36ns from performance counter `mcycle`
```
    li      a4, 0
    c.mv    a3, a4
```
- test2, set the value of `rs2` to be non-zero
we observe 24ns
```
    li      a4, 2047
    c.mv    a3, a4
```

PoC
These instructions violate the RISC-V Zkt constant instruction list and result in CRET CVE security advisory.

Impact
These instructions create direct timing channels and will threat the security of sensitive parameters.

zarubaf, 2023-10-23:
`Zkte` is currently not implemented, but it is planned as part of the CVA6 evolution.

Pallavi-Borkar, 2024-10-01:
> Hello!
> 
> I have been trying to reproduce the CT violations and have yet not observed the described behaviour. @jooohndoe Could you please clarify the following:
> 
> * How  did you use the MCYCLE CSR to perform the measurements? Specifically, when is MCYCLES cleared and when is it read?
> * How did you compile the test cases, i.e. what optimisation flags did you use? This might be relevant since you are using the pseudo instruction "li", which might be expanded differently, depending on the immediate and the compiler settings.
> 
> It would help if you could provide the disassembly of the generated executable. Thanks!

Thank you for your interest in our work. Please find below the responses to your queries:
1. We clear the MCYCLES register just before running the compressed instruction and read it just after the instruction.

2. The compiler flags are as follows: (-mcmodel=medany -l -std=gnu99 -O0 -g -fno-common -fno-builtin-printf -Wall -Wno-unused-function -Wno-unused-variable)

Please check the attached sample test cases for the same (the compressed instruction is on line 18). We have also replied to your email query regarding this in detail.

[compressed.zip](https://github.com/user-attachments/files/17210139/compressed.zip)

TobiasKovats, 2024-10-02:
Hello!
Thank you for your response and the provided test cases. When compiling them with the provided flags, I obtain the following disassembly (I have removed the printf):
```
    80000000:	7139                	c.addi16sp	sp,-64
    80000002:	fc22                	c.sdsp	s0,56(sp)
    80000004:	0080                	c.addi4spn	s0,sp,64
    80000006:	fe043423          	sd	zero,-24(s0)
    8000000a:	fe043023          	sd	zero,-32(s0)
    8000000e:	fc043c23          	sd	zero,-40(s0)
    80000012:	fc043823          	sd	zero,-48(s0)
    80000016:	47c1                	c.li	a5,16
    80000018:	fcf43423          	sd	a5,-56(s0)
    8000001c:	fc043023          	sd	zero,-64(s0) # Until here the stack is set up for the local variables.
    80000020:	7ff00713          	addi	a4,zero,2047 # The leaking immediate is loaded. In the other test case, this is a c.li	a4,0 and the next instruction is stored at 80000022.
    80000024:	4881                	c.li	a7,0
    80000026:	b00027f3          	csrrs	a5,mcycle,zero # The MCYCLE CSR is read once.
    8000002a:	fef43423          	sd	a5,-24(s0) # The result is written to the stack.
    8000002e:	9e3a                	c.add	t3,a4 # The leaking instruction is executed.
    80000030:	b00027f3          	csrrs	a5,mcycle,zero # The  MCYCLE CSR is read again.
    80000034:	fef43023          	sd	a5,-32(s0) # The result is written to the stack.
    80000038:	fe043703          	ld	a4,-32(s0) # The results are retrieved and the diff is computed.
    8000003c:	fe843783          	ld	a5,-24(s0)
    80000040:	40f707b3          	sub	a5,a4,a5
```
Using these test cases, I measure a timing difference of one cycle: 9 cycles for the immediate of 0, 8 cycles for 2047. Could you please confirm if this matches your observations?

Thank you!

Phantom1003, 2024-11-18:
I noticed that the two code snippets are not the same length, have you guys tried to align the code?
Maybe that's why you find a CT violation? 😂

```
   0:   4701                    li      a4,0
   2:   4881                    li      a7,0
   4:   b00022f3                csrr    t0,mcycle
   8:   9e3a                    add     t3,t3,a4
   a:   b0002373                csrr    t1,mcycle

   0:   7ff00713                li      a4,2047
   4:   4881                    li      a7,0
   6:   b00022f3                csrr    t0,mcycle
   a:   9e3a                    add     t3,t3,a4
   c:   b0002373                csrr    t1,mcycle

```