openhwfoundation/cva6 #1557: [TASK] investigate possible FPGA build time optimizations

cathales, 2023-10-20:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

https://github.com/openhwgroup/cva6/pull/1535 reduced the number of jobs run for each PR push. The only remaining long job is FPGA build (38 minutes). For comparison the second longest job is ASIC syntesis (8 minutes).

Reducing FPGA build time would be really interesting to make the whole pipeline shorter and to merge PRs faster.

### Required Changes

A possible solution could be caching built Vivado IPs

### Current Status

To be investigated

### KPI (KEY Performance Indicators)

FPGA build job duration

### Description of Done

TBD

cathales, 2023-10-20:
@JeanRochCoulon can you assign this issue’s project to CVA6, please?

jquevremont, 2023-11-21:
@sjthales encounters similar FPGA build time.