openhwfoundation/cva6 #1559: [TASK] Add Architectural tests for Atomic extensions

fatimasaleem, 2023-10-20:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

Recently atomic arch-tests have been upstreamed by 10x team. Need to update the CVA6 arch-tests yaml files to add those tests.

### Required Changes

Update the hash for install arch-tests
Update yaml files for cv32 step 1 configuration and cv64 for adding atomic tests

### Current Status

atomic arch-tests are not included in the CVA6 regressions

### Description of Done

Merged PR

closed 2023-10-23 by fatimasaleem

JeanRochCoulon, 2023-11-30:
@AyoubJalali  Can you confirm these tests have been added to the CI ? If you think the task is complete, can I ask you to comment and update the status from "done" to "finished" ?

AyoubJalali, 2023-11-30:
> @AyoubJalali Can you confirm these tests have been added to the CI ? If you think the task is complete, can I ask you to comment and update the status from "done" to "finished" ?

A extension isn't supported by the embedded configuration, that's right ?