openhwfoundation/cva6 #1565: [DECODER] : Missing condition on the XLEN for RV64B instruction

AyoubJalali, 2023-10-23:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
I notice the in the [decoder.sv](https://github.com/openhwgroup/cva6/blob/master/core/decoder.sv), there's some RV64b instructions (bitmanip 64 instructions) don't have the famous condition -> if (XLEN == 64), normally these instruction can't be decoded only if XLEN = 64.
I'm talking about these 2 blocks of code maybe there's more : 
[from ](https://github.com/openhwgroup/cva6/blob/38b1da26c3597881bbf543894322df7dcefc7a0e/core/decoder.sv#L651)
[to ](https://github.com/openhwgroup/cva6/blob/38b1da26c3597881bbf543894322df7dcefc7a0e/core/decoder.sv#L660) except ZEXTH it's RV32b instruction.
also [from ](https://github.com/openhwgroup/cva6/blob/38b1da26c3597881bbf543894322df7dcefc7a0e/core/decoder.sv#L755)
[to ](https://github.com/openhwgroup/cva6/blob/38b1da26c3597881bbf543894322df7dcefc7a0e/core/decoder.sv#L764)

AyoubJalali, 2023-10-23:
So I think we should only add this condition in the ALU.sv, and also remove zext.h from RV64b instruction

JeanRochCoulon, 2023-10-24:
Hello @fatimasaleem 
Ayoub has found an issue in BitManip implementation, can you confirm it and possibly propose a PR ?

closed 2023-11-01 by fatimasaleem