openhwfoundation/cva6 #1569: [TASK] Support only cva6.py commands to simulate cva6

JeanRochCoulon, 2023-10-24:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

Today two set of commands are available to simulate cva6: cva6.py coming from riscv-dv repo and "make verilate" supported by Makefile. To reduce confusion for newcomers and increase the simulation env quality, we will support only the cva6.py commands and remove the "make verilate" commands from Makefile.

### Required Changes

Move Github action CI to cva6.py commands

### Current Status

Today two set of commands are available to simulate cva6

### Risks

Users using "make verialte" should change their habits.

### Prerequisites

Before removing make verilate, 
- GitHub actions CI must be move to cva6.py commands and 
- we must confirm all current features are supported by cva6.py

### Description of Done

"make verilate" removed from Makefile

JeanRochCoulon, 2023-11-03:
@ValentinThomazic

valentinThomazic, 2023-11-30:
Usage of the proxy kernel which was still using Makefile targets to run simulation has been dropped.
See : https://github.com/openhwgroup/cva6/pull/1663
This means that, at this point, all simulation features should be supported by cva.py

JeanRochCoulon, 2023-11-30:
GitHub Actions uses "make verilate" to simulate. A possible action would be to switch this CI from "make verilate" to "cva6.py".     The ideal action would be to adapt Github Actions to "cva6.py" and to remove the relative "make verilate" commands from the Makefile.  I wonder what is the plan for you @MarioOpenHWGroup and @MikeOpenHWGroup ?

JeanRochCoulon, 2024-03-01:
Good job. The task is finished.

closed 2024-03-01 by JeanRochCoulon