openhwfoundation/cva6 #1570: [Question] Synthesis error

mahdi259, 2023-10-24:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi
My target was implementing CVA6 on FPGA. I run `make clean` and `make fpga` and I got error. The output of command line and vivado log is attached. 
[error.txt](https://github.com/openhwgroup/cva6/files/13110947/error.txt)
[hs_err_pid42902.log](https://github.com/openhwgroup/cva6/files/13110948/hs_err_pid42902.log)

mahdi259, 2023-10-24:
I opened ariane.xpr now and run synthesis. Here is the picture of vivado.
[image]

JeanRochCoulon, 2023-10-24:
@Gchauvon    Do you have an idea ?

mahdi259, 2023-10-24:
The problem was missing packages.
I run
 `sudo apt install libcanberra-gtk-module libcanberra-gtk3-module`
And was synthesized successfully.
Thanks.
[image]

closed 2023-10-24 by mahdi259