openhwfoundation/cva6 #1573: [TASK] Cascade CPU fuzzing research bugs track

acaldaya, 2023-10-25:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Recently the paper: _Cascade: CPU Fuzzing via Intricate Program Generation_ showcased some bugs of the CVA6. The goal of this issue is to track which bugs have been fixed and the corresponding PR. 

|ID (paper) | Description| CVE|Status|
|:--:|:--|:--:|:--:|
|C1| Double-precision multiplications yield wrong sign when rounding down |2023-34904||
|C2| Single-precision floating-point operations may treat NaNs as zeros |2023-34906||
|C3| Division by NAN incorrectly sets NX and NV fflags |2023-34905||
|C4| The inexact (NX) flag not set in case of overflow or underflow  |2023-34907||
|C5| Division of zero by zero incorrectly sets the DZ flag |2023-34909||
|C6| Plus and Minus infinity microarchitectural structures are inverted |2023-34910||
|C7| Infinities are not rounded properly and stick to infinity |2023-34911||
|C8| Spurious exceptions when reading some performance counters  |2023-34911||
|C9| Wrong supervisor performance counter access control |2023-42311||
|C10| Under some microarchitectural circumstances, wrong NAN conversion | 2023-34908 ||

### Required Changes

When a bug of the table is fixed, please link this issue with the corresponding bug ID or CVE in the PR.

### Current Status

It is not straightforward to know if the bugs are already fixed.

### Description of Done

When all bugs have been fixed and the status column is updated with the corresponding PR.

JeanRochCoulon, 2023-10-26:
Thank you for the work done on CVA6 !!   I like it!  
Maybe I ask you the corresponding CVA6 hash on which the verification has been executed ?

JeanRochCoulon, 2023-10-26:
@acaldaya  Would you be available to present the results in cva6 verification meeting ?  it is scheduled at 2pm CET on Thursday

@ASintzoff moderates it

acaldaya, 2023-10-26:
Sorry for the confusion, it is not my research, but really interested.

I will write to the authors and let them know about this thread. According
to the paper, a C1 fix was merged, but I didnt find a ref of it.

JeanRochCoulon, 2023-10-26:
@flaviens Would you be available to present the work results ?

suppamax, 2023-10-26:
I have found these existing issues, which seem to be related to the report
https://github.com/openhwgroup/cva6/issues/1213
https://github.com/openhwgroup/cva6/issues/1212
https://github.com/openhwgroup/cva6/issues/1192
https://github.com/openhwgroup/cva6/issues/1091
https://github.com/openhwgroup/cva6/issues/1071

flaviens, 2023-10-26:
Hello, thank you for your interest!

I would like to underline that all the bugs were reported to you many months before the paper went public, in accordance with responsible disclosure.

Here are some links to help you.

https://github.com/openhwgroup/cvfpu/issues/72
https://github.com/openhwgroup/cva6/issues/1071
https://github.com/openhwgroup/cva6/issues/1192
https://github.com/pulp-platform/fpu_div_sqrt_mvp/issues/15
https://github.com/pulp-platform/fpu_div_sqrt_mvp/issues/17
https://github.com/pulp-platform/fpu_div_sqrt_mvp/pull/20/commits/ecfabd1f37e4354ead96bf347ab648addde42491
https://github.com/pulp-platform/fpu_div_sqrt_mvp/pull/20/commits/de10d674cbb2da026aeb1995a3fb45c2623a3761
https://github.com/openhwgroup/cva6/issues/1212
https://github.com/openhwgroup/cva6/issues/1213
https://github.com/openhwgroup/cvfpu/issues/77

https://github.com/openhwgroup/cvfpu/pull/73
https://github.com/pulp-platform/fpu_div_sqrt_mvp/pull/16
https://github.com/pulp-platform/fpu_div_sqrt_mvp/pull/18
https://github.com/pulp-platform/fpu_div_sqrt_mvp/pull/20

Regarding the presentation, I must unfortunately decline for this specific time slot (this is a very short notice!).
@JeanRochCoulon, if you would like a presentation or discussion, please contact me by email directly to schedule it, I'd be very glad to have the opportunity to exchange with you.

Kind regards,
Flavien

flaviens, 2023-10-26:
Additionally, @KatCe suggested that you may be interested in [this issue (Y1)](https://github.com/YosysHQ/yosys/issues/3773) as well, despite being seemingly a yosys-related issue.