openhwfoundation/cva6 #1586: [TASK] Add extension and features tags on tests in testlists

Gchauvon, 2023-10-31:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

 Add tags on tests in testlists to identify extension and/or micro-architectural features required by CVA6 to run the tagged test.
There is an example in [testlist_hwconfig.yaml](https://github.com/openhwgroup/cva6/blob/master/verif/tests/testlist_hwconfig.yaml)

```
- test: rv32ui-v-add
  iterations: 1
  needs:
    - xlen: 32
    - fpu: 1
    - c_ext: 1
  path_var: TESTS_PATH
  gcc_opts: "-DENTROPY=0x1 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -I<path_var>/riscv-tests/isa/macros/scalar/ <path_var>/riscv-tests/env/v/entry.S <path_var>/riscv-tests/env/v/vm.c <path_var>/riscv-tests/env/v/string.c -I<path_var>/riscv-tests/env/v/ -I<path_var>/riscv-tests/riscv-target/spike/"
  asm_tests: <path_var>/riscv-tests/isa/rv32ui/add.S
```

### Required Changes

- Keep only one testlist for each regression ( riscv-tests, riscv-compliance, riscv-arch-test, ...)
- Tag all tests with extension and/or micro-architectural features in CVA6 required to run the specific test
- Compare the list of required extension and/or micro-architectural features with the configuration of the CVA6 used to run the test ( core/include/*_config_pkg.sv )
- CVA6 should only run tests that are compatible with its configuration.

### Current Status

An idea of what is expected is written in  [testlist_hwconfig.yaml](https://github.com/openhwgroup/cva6/blob/master/verif/tests/testlist_hwconfig.yaml), [config_pkg_generator](https://github.com/openhwgroup/cva6/blob/d92f0f76d053182c7ab41bc94bcf1a61e9757b00/util/config_pkg_generator.py) and [cva6.py](https://github.com/openhwgroup/cva6/blob/d92f0f76d053182c7ab41bc94bcf1a61e9757b00/verif/sim/cva6.py).

### Description of Done

Existing testlists are removed and replaced with tagged one.

JeanRochCoulon, 2023-10-31:
Related to #1569

JeanRochCoulon, 2024-05-19:
@valentinThomazic will address it on July...