openhwfoundation/cva6 #1588: [Decoder] ZEXTH is a RV32 instruction

AyoubJalali, 2023-11-01:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, I notice that the ZEXTH instruction is still with the RV64 instruction block (I think you missed to remove it), and I see it in 3 different place in the [decoder.sv](https://github.com/openhwgroup/cva6/blob/master/core/decoder.sv) : 

1. [First one](https://github.com/openhwgroup/cva6/blob/f301d6967517336a21a58c9b8a00ea3186906c01/core/decoder.sv#L596)
2. [Second ](https://github.com/openhwgroup/cva6/blob/f301d6967517336a21a58c9b8a00ea3186906c01/core/decoder.sv#L639)
3. [Third one](https://github.com/openhwgroup/cva6/blob/f301d6967517336a21a58c9b8a00ea3186906c01/core/decoder.sv#L671)

But the real issue here I think is [here](https://github.com/openhwgroup/cva6/blob/f301d6967517336a21a58c9b8a00ea3186906c01/core/decoder.sv#L681), because some tests failed, and i suspect this line to be the problem.

JeanRochCoulon, 2023-11-08:
@fatimasaleem  Can we close this GitHub Issue after the PR merging ?

JeanRochCoulon, 2023-11-10:
@AyoubJalali Can you close this issue (indicate the PR number please) ?

AyoubJalali, 2023-11-13:
fix by : PR #1579

closed 2023-11-13 by AyoubJalali