openhwfoundation/cva6 #1592: Information about the division RTL module : serdiv.sv

TulikaSi, 2023-11-03:
The duration of the division is data-dependent and to perform the division operation always in constant time we need to modify the RTL. We can either take the worst case and make the division always take 64 cycles under all scenarios or come up with some other alternative.

I was trying to understand the algorithm implemented for the **division module(serdiv.sv)**. If I can get some information about the algorithm that is implemented in the RTL it would be really helpful.

Thanks

MikeOpenHWGroup, 2023-11-07:
Good question @TulikaSi.  The divider used by the FPU in CVA6 is a `serial divider`.  This architecture is very area efficient, but the cycle-time of operations is dependent on the data and the worst case is 64 clock cycles.  Changing this would require a significant amount of work which is not currently being considered by any OpenHW project.

I have attached a set of slides that is a good explanation of this architecture.  Keep in mind that the document is _not_ the micro-architecture of the CVA6 FPU, It is attached here for informational purposes only.

You should also be aware that:
- The FPU in the CVA6 is very out-of-date and is known to have many issues.  This is not a big concern at this point as none of the "production devices" derived from CVA6 support the F or D ISA.
- The OpenHW Group is encouraging all projects that need the F or D ISA to use [CVFPU](https://github.com/openhwgroup/cvfpu).  This is IP started out as `fpnew` at ETHZ.  It is used by the CV32E40P v2.0.0 and has many bug fixes for the F ISA (64-bit versions of CVA6 will need to verify the D ISA).  In theory, integrating latest CVFPU into CVA6 will not be difficult, but as far as I know, nobody has investigated this.

I hope this answers your question.  If it does, please close this Issue.

[DividerBackground.pdf](https://github.com/openhwgroup/cva6/files/13281119/DividerBackground.pdf)

pascalgouedo, 2023-11-07:
Hi @Mike

Not sure the original question was concerning Floating-Point Division...
serdiv.sv is nowhere in CVFPU so I guess it is the 64-bit integer division @TulikaSi is talking about.

Just my 2 cents...

Moschn, 2023-11-07:
Why do you want the divison instruction to take constant time? For secure constant time implementations, the division instruction is usually avoided entirely. For example, the [RISC-V zkt](https://github.com/rvkrypto/riscv-zkt-list/blob/main/zkt-list.adoc#rvm-multiply) excludes division explicitly. Or do you need this for a real-time system where hard deadlines are important?

MikeOpenHWGroup, 2023-11-08:
Good question @Moschn.  Keep in mind that the CVA6 is not specifically targeting secure applications and does not implement the zkt ISA. To date, the CV32E40S is the only CORE-V "secure core", although I strongly suspect it will not be the last.

This issue is not a place to debate this.  If there is interest in this topic the [Discussion board](https://github.com/openhwgroup/cva6/discussions) is the best place for that.

@TulikaSi, has your question been answered?  If so, please close this issue.  Thanks.

TulikaSi, 2023-11-09:
Hi @Moschn @MikeOpenHWGroup 
Thanks for the explanation but my question concerning the division operation to take a constant time came up based on the 
 **ISA-110 of the requirement specification** (https://github.com/openhwgroup/cva6/blob/master/docs/02_cva6_requirements/cva6_requirements_specification.rst#risc-v-standard-instructions)
The Note to ISA-110 states that **In the current design, the duration of the division is data-dependent, which can be a security issue** so now the question in my mind is if we are not hinting toward constant time implementation of division instruction so what does this requirement demands.

zarubaf, 2023-11-09:
The ISA requirement was written before `Zkt` was proposed - I am in favor of re-discussing the requirement and follow the upstream path. Or at least make the implementation switchable in case that is a hard requirement for the Thales use-case. Making it constant time means that you would need to pad all divisions to worst case time (32 or 64-bit).

Moschn, 2023-11-21:
FYI, there are open issues that indicate CVA6 is currently _not_ compliant with Zkt (#1547). This also might just be the tip of the iceberg, as we do not know if more instructions that are supposed to be constant-time exhibit a variable timing behavior for some edge cases. 

Since this can be security-critical, we should be careful about how we "claim" compliance with Zkt. Stating that the core is compliant could raise a false sense of security, as no verification of the core in this respect has been performed (AFAIK).

jquevremont, 2023-11-21:
> FYI, there are open issues that indicate CVA6 is currently _not_ compliant with Zkt (#1547). This also might just be the tip of the iceberg, as we do not know if more instructions that are supposed [...]

Acknowledged.

jquevremont, 2023-12-04:
> Today's meeting: security applications do not use integer division and have no need for this feature. @jquevremont to check with safety team if this requirement is still relevant.

Answer: there are other sources of unpredictability in the core. It makes no sense to remove this one in particular. 

Conclusion: we can drop the fixed-time division from the requirement specification.