openhwfoundation/cva6 #1597: [BUG] cva6/core/Flist.cva6 mentions " ENV variable CVA6_REPO_DIR" but there are others

rajatkmitra, 2023-11-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The header should also mention that HPDCACHE_DIR and HPDCACHE_TARGET_CFG need to be set. I noticed that HPCACHE_DIR can easily be inferred as it already resides under -
cva6/core/cache_subsystem/hpdcache/
So why do we need a seperate macro for it ? Also where do I find the options for HPDCACHE_TARGET_CFG?

MikeOpenHWGroup, 2023-11-06:
Hi @rajatkmitra, thanks for your interest in OpenHW and CVA6.  We are updating our script-ware and associated documentation to simplify supporting CVA6 and also to enable compilation of multiple configurations of the CVA6.  We will ensure your questions are addressed in a future update of the docs.

rajatkmitra, 2023-11-06:
Many thanks @MikeOpenHWGroup  !!
--Raj

rajatkmitra, 2023-11-07:
Hi @JeanRochCoulon 
As you noted in the othere post #1358 , I had some difficulties using pk.. The instructions were not clear. But  @valentinThomazic
clarified the directions. I noticed that the default Makefile is looking for verilator target in cva6/tmp. Again this was missing from the instructions. I have still to try using pk and will inform shortly if that was successful. My goal is to replace verilator with Xcelium and then use pk + Xcelium. If that is successfult I plan to run ariane SoC on emulator - Z2.

JeanRochCoulon, 2023-12-08:
Hello @rajatkmitra 
@valentinThomazic has just updated the README.md.  Pk support has been removed but recommandations to simulate CVA6 updated. Can you review it to give us your feedback ?

rajatkmitra, 2023-12-09:
Hi @JeanRochCoulon 
Apologies for the delay.. Yes I will try this and inform on Monday

rajatkmitra, 2024-01-09:
Yes it's okay to close... Following up with @valentinThomazic  on a separate thread...

closed 2024-01-09 by JeanRochCoulon