openhwfoundation/cva6 #1598: *** FAILED *** When executing simple addition

Midas-sudo, 2023-11-05:
After setting everything up I ran a small hello world in the CVA6 without any issue, with the phrase "Hello from CVA6!" displaying correctly and being able to analyze the waveforms. However when i tried something a little different, see code below, the simulation starts, idles for a long time, and then fails with:
[image]

The code used is a simple addition, I was trying to see if I would be able to understand  the flow of instructions of this simple program in the wave forms:
```c
#include <stdio.h>

int main(int argc, char const *argv[]){
    int a, b, c;
    a = 5;
    b = 2;

    c = a + b;
    printf("%d\n", c);
    return 0;
}
```

I really dont understand what I'm doing wrong. I noticed in the signal of the instruction seems to get stuck on instruction 0xFEE7BC23 until the simulation eventually time-outs.
[image]

MikeOpenHWGroup, 2023-11-06:
Hi @Midas-sudo, thanks for your interest in OpenHW and the CVA6.  I have a request and a comment:
- Request: please update this issue with the _exact_ commands you typed, starting with `git clone...`.
- Comment: your test-program looks like something that you would reasonably expect to run on a complete Linux system.  The execution environment of the CVA6 in simulation is nothing like that.  I am not the expert on this, but I doubt that arguments to `main()` are supported and depending on your setup, calls to `printf()` may not work as expected.  Have a look at other test-programs and follow their example.

Midas-sudo, 2023-11-06:
@MikeOpenHWGroup thank you for your response! I thought it should be able to run (even though I see that this is not Linux machine). 
The reason that led me to believe was the fact that I can run this:
```c
#include <stdio.h>
int main(int argc, char const *argv[]) {
    printf("Hello World Test!\n");
    return 0;
}
```

Which is not that different. I'm currently not at home but as soon as I get home I will comment my journey of commands from the start ( not too sure if I will be able to remember every single one but I will try)

Also I've been compiling the c source code with riscv64-unknowne-gcc having as output an elf file (don't know if it helps).

I can't seem to find good examples of tests, and the ones I found I can't seem to understand.

I'm supposed to use CVA6 for my thesis, requested by my orientation teachers. I've asked for their help but they are a bit busy all the time and I'm starting to stress out... I was supposed to implement a new Fu and instructions in the CVA6 but I can't seem to overcome the challenge that has become installing the necessary "tools"

Thank you for your time ❤️

Midas-sudo, 2023-11-08:
@MikeOpenHWGroup Sorry for the late response I'm in the middle of exams so my days have been a bit chaotic.

So I've tried multiple approaches to install/run stuff. I've done it using the scripts contained on the cva6 repo, and I've done it using the dedicated repos for each part (riscv-toolchain, spike, pk etc).

Currently my installation is all from CVA6 repo. My steps where:

```
git clone https://github.com/openhwgroup/cva6.git
cd cva6
git submodule update --init --recursive

export RISCV=my_riscv_dir

cd util/gcc-toolchain-builder
bash ./get-toolchain.sh

# For the build prerequisites, see the local README.md.

# Build and install the GCC toolchain.
bash ./build-toolchain.sh $RISCV

# Return to the toplevel CVA6 directory.
cd -

export DV_SIMULATORS=veri-testharness,spike
bash verif/regress/smoke-tests.sh
```

And it seems to install everything correctly. 

Since my initial post, I've discovered/understood that you are supposed to run tests and I guess user-applications through the cva6.py right?

Doing it with it I managed to successfully run the hello_world that comes with the test, and it has some additions so that's working. Now I tried to add new instructions to the decoder (the next step of my journey but the run simply times out)

I know it correctly added the instruction to the riscv-toolchain because running it only using spike it works correctly. Cant seem to understand what I'm doing wrong now.

I cant seem to understand where the verification of the opcode of the instruction is made.

MikeOpenHWGroup, 2023-11-09:
Good to see you are making progress @Midas-sudo.

> I cant seem to understand where the verification of the opcode of the instruction is made.

Currently, the test-program is run twice: once through the RTL and once more through the Spike model.  Both tests create instruction trace files and these are compared in a post-processing step.   We will be moving very shortly to a single simulation that executes the RTL and Spike simutaneously and the trace is compared as each instruction is retired.  This eliminates the need for a post-processing step.   We call this "tandem mode" and it will be available very soon (maybe today).

I think you are on your way, so I'll close this issue now.

closed 2023-11-09 by MikeOpenHWGroup