openhwfoundation/cva6 #1612: Seeking for guidance in CVA6 core only simulation

AnikBalo, 2023-11-09:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi, Is it possible to simulate CVA6 core only with the test program like (Dhrystone, etc.) with simple testbench? Is there any procedure/guideline/methods?

MikeOpenHWGroup, 2023-11-09:
Hi @AnikBalo, thanks for your interest in the CVA6.  Currently, our scriptware is designed to run suites of tests, not individual tests.  The easiest to follow example is the [smoke-test](https://github.com/openhwgroup/cva6/blob/3c455109348acc6061109897075af29658c37e86/verif/regress/smoke-tests.sh#L1).

My advice would be to follow the README to get the smoke-test running on and then looking into the script to see how it lanuches individual tests.

FYI, we have plans to update our scriptware to support running individual tests, similar to what we have for our embedded cores.

If this answers your question, please close this Issue - thanks!

AnikBalo, 2023-11-09:
Thanks a lot.

AnikBalo, 2023-11-09:
Thanks a lot!

closed 2023-11-09 by AnikBalo