openhwfoundation/cva6 #1613: [DECODER] : Separate between RV32 & RV64 instruction

AyoubJalali, 2023-11-09:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Still on the same issue, in the [line](https://github.com/openhwgroup/cva6/blob/3c455109348acc6061109897075af29658c37e86/core/decoder.sv#L725), REV8 has 2 instr[31:20] value to be valid instrcution, but in fact REV8 is a RV64 instruction if _**instr.instr[31:20] == 12'b011010111000**_ & it's a RV32 if **_instr.instr[31:20] == 12'b011010011000_**, so it sould add a condition on this to improve CC

JeanRochCoulon, 2023-11-10:
Hello @AyoubJalali    A dediccared PR has been merged concerning this issue. If the issue is fixed, please close it (by indicating the PR number).

ASintzoff, 2023-11-13:
No change (except the verible formatting) has been done on that line since the introduction of bitmap RTL code.

AyoubJalali, 2023-11-13:
> Hello @AyoubJalali A dediccared PR has been merged concerning this issue. If the issue is fixed, please close it (by indicating the PR number).

That's a new one

JeanRochCoulon, 2023-12-17:
@AyoubJalali   #1714 should have fixed this issue. If you confirm, you would have the privilege to close the issue ;-)

closed 2023-12-18 by AyoubJalali