openhwfoundation/cva6 #1619: Getting assertion error during simulation of cva6 using Questasim

UsamaEmam, 2023-11-10:
Getting assertion error during simulation of the latest version of cva6 on the master using Questasim 
The failing assertion in core/pmp/src/pmp.sv :
[image]

The assertion fails only once during simulation but the test finishes and triggered the following UVM INFO: 
[image]

Is this assertion accepted? or how to avoid such scenario?

MikeOpenHWGroup, 2023-11-13:
Hi @UsamaEmam, thanks for your issue.  Can you update this issue with the _exact_ command-lines you used starting with `git clone....`?  This will be a big help to debug the issue.  Also, please let us know the version of Questasim.

Hi @zarubaf, I am assigning this to you as the original author (Moritz Schneider <scmoritz@student.ethz.ch>) is probably no longer active on CVA6.

zarubaf, 2023-11-14:
Yeah, this is an intermittent assertions that fires before reset, so can be ignored. But should be fixed for a cleaner flow.

closed 2023-11-14 by zarubaf