openhwfoundation/cva6 #1624: Immediate assertions in the CVA6 RTL

MikeOpenHWGroup, 2023-11-13:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

This issue is motivated by #1619.

A quick grep of the CVA6 RTL reveals approximately 50 `assert` statements of the form `assert (expression);`. These are called "_immediate assertions_" in the SystemVerilog LRM.  Immediate assertions without an `else` clause are dangerous because they will only emit a simulator-specific message to stdout, without interacting with the simulator or the testbench.  That is, once the assertion fires and the message is written to stdout, there is no further record of the assertion, so it is not possible for the testbench to know that the assertion happened, and bugs can be missed.

For that reason there should be an _else_ clause for all assertions:
```
assert (expression)
  else `uvm_error("<Identifier>", "Useful (?) error message")
```
In non-UVM environments. the `uvm_error()` could be replaced with either $error() or $fatal().  In these cases, the script that launches the simulation must post-process the logfile to check for these messages.  This is not considered ideal since it is easy for the error message in the logfile to be missed.

To resolve this Issue I propose the following:
- Create a "_cvv_error()_" macro that will emit either a `uvm_error()` for UVM simulations or `$fatal()` for non-UVM sims.
- Update all immediate assertions to add an `else` clause as follows:
```
assert(expression)
  else `cvv_error("<Identifier>", "Useful (?) error message")
```

JeanRochCoulon, 2025-01-02:
@AnouarZajni  I assigned you to this issue because the point need to be discussed. Fix or not fix, that is the question.

AnouarZajni, 2025-01-08:
Hi @MikeOpenHWGroup,
Yes immediate assertions without else $error/$fatal will printout to stdout but also will change the simulator exit status to "1". Unfortunately cva6.py is not catching any exit status and hence can make the test PASS even with an assert failure.
it's true that using 'else `cvv_error("<Identifier>", "Useful (?) error message")' will solve partially the problem (Different macro for RTL and UVM) but this will generate many file modifications, I will suggest to update cva6.py to catch the exit code instead which will be also useful for any other errors like OS ones.

Regards,
Anouar

JeanRochCoulon, 2025-01-24:
@MikeOpenHWGroup   After another discussion with @AnouarZajni we propose to add an "else" with "$error()". This allows to get a generic RTL which does not depend on testbench. The simulation can stop when assertion is triggered.
Tell us what you think.
This fix will be a good first RTL modification for @valentinThomazic  ;-)

AnouarZajni, 2025-01-27:
Thanks @MikeOpenHWGroup  for the clarifications. Yes having both "assert else" and simulation exit code will be perfect.
However regarding the macro, I checked that most of the "assert without else" are in RTL code so in an UVM TB we will end up by having uvm errors inside non-uvm modules which is a bit confusing.

khandelwaltanuj, 2025-02-06:
Hello all,

This makes me think of a perl script( (https://github.com/openhwgroup/cv-hpdcache-verif/tree/main/scripts/scan_logs.pl)  that we have developped to avoid false positives and this is the part our flow to run this script after every run. It simply scans the log and return pass or fail status. Here is a snippet from the help of this script 

"
_**- It is essential not to produce false positives. For this reason, this script
will always check for an explicit message indicating that the job has
completed. This helps ensure that a false positive is not produced if
a truncated (or potentially empty) log file is provided as input.

The script has a set of default rules which are used to identify warnings,
errors and successful test completion. These rules can be customized using
configuration files. The user specifies the name of the pattern configuration
file with the '-pat' command line option."**_

Ideally it should have been in the cv_dv_utils but it is not. Though there is a use case in : https://github.com/openhwgroup/cv-hpdcache-verif/blob/main/scripts/sim/run_reg.py. 

It can be run on one log or multiple logs at a time. 

I hope it will be of some help.

Regards
Tanuj Khandelwal

MikeOpenHWGroup, 2025-03-06:
Sorry for my late reply @AnouarZajni.

> However regarding the macro, I checked that most of the "assert without else" are in RTL code so in an UVM TB we will end up by having uvm errors inside non-uvm modules which is a bit confusing.

The idea is that in the RTL we add an `else` clause to all immediate assertions using the macro above:
```
assert(expression) else `ASSERT_ERROR(useful_name)
```
So there is no confusion - a non-UVM testbench will use $error() and a UVM testbench will use `uvm_report_error()`.

The comments from @khandelwaltanuj are also useful to this discussion.  I recommend adding the "else clause" to all immediate assertions in the RTL (using the above macro) **_and_** adding the log scanning capability of  something like [run_reg.py](https://github.com/openhwgroup/cv-hpdcache-verif/blob/main/scripts/sim/run_reg.py).

khandelwaltanuj, 2025-03-10:
Hello @MikeOpenHWGroup 

I believe, once I get going for cv64a60ax, I will integrate scan_logs in run_reg.py. 
If needed be we can arrange a presentation of scan_log which is developed by Adrian Evans of our team. 

thanks and regards
Tanuj Khandelwal

MikeOpenHWGroup, 2025-04-03:
Hi @valentinThomazic, the changed that were made in PR #2888 resolve my primary concern, so I will close this issue.  I did make a few review comments (after the merge), so please have a look at those.  You can decide later if you wish to create another PR to address those comments - they are minor, so I will not create an Issue for them.

commit 37049ee7 (2025-04-01): add macros for assertions else clauses (fix #1624) (#2888)
closed 2025-04-03 by MikeOpenHWGroup