openhwfoundation/cva6 #1625: [Spike] : Mismatch of address mapping between spike & RTL

AyoubJalali, 2023-11-13:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, 
So after the merge of spike tandem, one of the directed tests that I wrote (to cover some corner cases in the ISACOV) failed, in this case spike reach  his maximum steps, so after a quick debug, I found that spike raise an exception of loadaccess_fault on an address (example : 0xffff_fff0), in the opposite side the CVA6 does raise any exception and threat the address in his valid range.

[image]

So to reproduce the problem you only need to execute the command line :  
python3 cva6.py --testlist=../tests/testlist_isacov.yaml --test load_reg_hazard --iss_yaml cva6.yaml --target cv32a6_embedded --iss=vcs-uvm,spike (run it in cva6/verif/sim) of course using the new spike (cva6 HASH 5e68fc0a3fde3333d645d0b43fbce58bebabe6cf)

MarioOpenHWGroup, 2023-11-16:
Do we have a specification of the memory space address?

MarioOpenHWGroup, 2023-11-21:
Following the specifications, this address is out of both `ExecuteRegion` and `CachedRegion` . 

https://docs.openhwgroup.org/projects/cva6-user-manual/01_cva6_user/Parameters_Configuration.html#parameters

zchamski, 2023-11-21:
OK, so that's an RTL issue (access outside the mapped space should be detected).  @JeanRochCoulon, what's your view on this?

JeanRochCoulon, 2023-11-21:
@zchamski   To my mind, test is maybe wrongly compiled/linked. Why 0xfffffff0 address is acccessed ? Of course cva6 does not implement this address but the test should not use this address.

zchamski, 2023-11-22:
Looking at the code, the access is explicitly generated as `lw zero, -16(zero)`, so the compilation flow only generates the corresponding instruction.

zchamski, 2023-12-12:
> OK, so that's an RTL issue (access outside the mapped space should be detected). @JeanRochCoulon, what's your view on this?

Any news from the CVA6 spec and/or implementation side?

JeanRochCoulon, 2024-09-12:
As CVA65X does not implement PMA checkers, Spike need to be modified to be aligned with CVA65X behavior.

zchamski, 2025-01-09:
> For the Spike implementation we need a spec of behavior upon out-of-range memory accesses [...]

@JeanRochCoulon, @ASintzoff: Can you please provide such a spec and add it to the User Manual?  This will be the common reference for Spike and for RTL.

Unmapped address behavior (e.g., the value patterns read from out-of-range memory locations) should also be described in the "platform" part of the `riscv-config` spec.  I already added support for memory region definitions (see https://github.com/openhwgroup/cva6/pull/2411), so RV-config modifications will be simply incremental.

zchamski, 2025-01-23:
@ASintzoff @JeanRochCoulon: In addition to mapped memory base and size please provide the value that Spike should put in registers when loading from an unmapped address.  Typical candidates are `0x0` and `0ffffffff`, and they depend on the platform ==> the platform design document should clearly specify them.

Also, please specify the behavior upon fetching 32 bits from an address that is 16b before the base or the end of the mapped area: are the lower (resp. upper) 16bits set to memory default?

JeanRochCoulon, 2025-01-23:
CVA6 PMA is disable. That is why CVA6 does not take care of the platform memory mapping. I propose to align Spike on CVA6, that is to say, by not implementing PMA in Spike.
As conclusion, Spike need to support alll addresses, from 0x0 to 0xffffffff, and memory can contain zero value by default (as it is Today I think).

zchamski, 2025-03-06:
The unmapped memory behavior being defined and no unaligned access allowed, the Spike side can now be implemented ==> adding this to my task pipeline.

commit 542d1b5a (2025-03-28): Fix #1625: Allow unmapped memory accesses in Spike for cv32a60x/cv32a65x. (#2882)

Enable silent fetch/load of zeroes when accessing unmapped memory locations in Spike:
* Bump CVV to use Spike supporting unmapped accesses when core-specific parameter allow_unmapped_mem_access is set to True.
* Add allow_unmapped_mem_access: True field to core-specific section of CV32A60X and CV32A65X Spike Yaml files.