openhwfoundation/cva6 #1628: Outdated manifest files

cathales, 2023-11-15:
Hello,

`ariane.core` and `scr_files.yml` mention a `src/scoreboard.sv` file. However the `src/` directory was removed by 3ddf797e95923fd11113c8e443046105dfbf8843 (#725, merged on Sep 24, 2021). So these files seem to be outdated since more than two years now.

I could not find issues about these files in the repository. Are they used? Should we update them? Should we remove them?

JeanRochCoulon, 2023-11-20:
@cathales   I propose you submit a PR to fix it

cathales, 2023-11-20:
I close this issue after the following comment from @Jbalkind about PR https://github.com/openhwgroup/cva6/pull/1644#issuecomment-1819345451

> We use flist.ariane and ariane.core. I think the other file is used by bender or ESP or something but I don't know. Please stop trying to delete them every few months. Of course these files will be "outdated" because the committers keep accepting PRs which add/remove/move core files without updating these manifest files. By their nature, these files will always be somewhat out of date and need to be updated via PR whenever the downstream project needs to update its submodule pointer to HEAD.

closed 2023-11-20 by cathales