openhwfoundation/cva6 #1629: Wrong code in install-riscv-arch-test.sh

zhangkanqi, 2023-11-15:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

There are some wrong code in install-riscv-arch-test.sh as fllows:

Now:

```shell
if ! [ -n "$ARCH_TEST_REPO" ]; then
  ARCH_TEST_REPO=https://github.com/riscv-non-isa/riscv-arch-test
  ARCH_TEST_BRANCH=main
  ARCH_TEST_HASH="a5a49fc9f244192649e57fe61b4513d9bc39b1e3"
fi
```

But i think it sholud be:

```shell
if ! [ -n "$ARCH_TEST_REPO" ]; then
  ARCH_TEST_REPO=**"**https://github.com/riscv-non-isa/riscv-arch-test**.git"**
  ARCH_TEST_BRANCH=**"**main**"**
  ARCH_TEST_HASH="a5a49fc9f244192649e57fe61b4513d9bc39b1e3"
fi
```

JeanRochCoulon, 2023-11-20:
related to #1569

valentinThomazic, 2023-12-04:
Hey @zhangkanqi, the current code actually works but you could submit a PR if you would like to improve it.
If everything is OK, could you close your issue ?

zhangkanqi, 2023-12-05:
Ok. Sorry for forgetting to close this issue.

closed 2023-12-05 by zhangkanqi