openhwfoundation/cva6 #1631: Guidance to simulate CVA6

Juan-Gg, 2023-11-15:
I would like to simulate CVA6 with Verilator and I am in need of some guidance as I have little to no experience on these matters. First, Is this is the place to ask these questions or I should look elsewhere?.

I am working under Ubuntu 22.04.2. This is what I’ve done so far:
-	I installed the RISCV toolchain from here: https://github.com/riscv-collab/riscv-gnu-toolchain/releases, specifically: riscv64-elf-ubuntu-22.04-gcc-nightly-2023.11.08. I followed this instructions: https://github.com/stnolting/riscv-gcc-prebuilt
-	Then I set $RISCV to /opt/riscv and added /opt/riscv/bin to path.
-	I installed Spike and Riscv-PK and compiled and simulated a hello world following this video: https://www.youtube.com/watch?v=zZUtTplVHwE&ab_channel=DerryPratama
These are the gcc and toolchain versions:
    $ gcc --version
    gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
    Copyright (C) 2021 Free Software Foundation, Inc.
    $ riscv64-unknown-elf-gcc --version
    riscv64-unknown-elf-gcc () 13.2.0
    Copyright (C) 2023 Free Software Foundation, Inc.
I tried to follow the “Getting Started” instructions:
-	I cloned the CVA6 repository
-	I ran the following:
$export DV_SIMULATORS=veri-testharness,spike
$bash verif/regress/smoke-tests.sh
But it did not find verilator, so I added it to path by running:
$export PATH=$PATH:/home/username/Documents/cva6Repo/tools/verilator-v5.008/verilator/bin

Then I executed: $bash verif/regress/smoke-tests.sh and I don’t know how to interpret the output. I see some lines mentioning errors, such as:
`ERROR    ERROR return code: True/2, cmd:make veri-testharness target=cv32a60x`

The complete output is this:

<details>
  <summary><i>Complete output:</i></summary>

```
Building Verilator in /home/username/Documentos/cva6Repo/tools/verilator-v5.008/verilator...
Verilator will be installed in /home/username/Documentos/cva6Repo/tools/verilator-v5.008
VERILATOR_REPO=https://github.com/verilator/verilator.git
VERILATOR_BRANCH=master
VERILATOR_HASH=v5.008
VERILATOR_PATCH=/home/username/Documentos/cva6Repo/tools/../verif/regress/verilator-v5.patch
NUM_JOBS=4
M	docs/CONTRIBUTORS
M	include/verilated_types.h
HEAD está ahora en 21093fd1b Version bump
error: patch failed: docs/CONTRIBUTORS:154
error: docs/CONTRIBUTORS: patch does not apply
error: patch failed: include/verilated_types.h:1012
error: include/verilated_types.h: patch does not apply
configuring for Verilator 5.008 2023-03-04
checking whether to perform partial static linking of Verilator binary... yes
checking whether to use tcmalloc... check
checking whether to use -m32... no
checking whether to build for coverage collection... no
checking whether to use hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking for a BSD-compatible install... /usr/bin/install -c
compiler is g++ --version = g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
checking that C++ compiler can compile simple program... yes
checking for ar... ar
checking for perl... /usr/bin/perl
[... 458 lines cut ...]

Wed, 15 Nov 2023 13:23:46 ERROR    ERROR return code: True/1, cmd:/opt/riscv/bin/riscv64-unknown-elf-gcc -mcmodel=medany -nostdlib          -nostartfiles ../tests/custom/hello_world/hello_world.c          -I/home/username/Documentos/cva6Repo/verif/sim/dv/user_extension           -T../tests/custom/common/test.ld -g ../tests/custom/common/syscalls.c ../tests/custom/common/crt.S -lgcc -I../tests/custom/env -I../tests/custom/common -o /home/username/Documentos/cva6Repo/verif/sim/out_2023-11-15/directed_c_tests/hello_world.o  -march=rv32imac_zba_zbb_zbs_zbc_zicsr_zifencei -mabi=ilp32
make: entering directory '/home/username/Documentos/cva6Repo'
Makefile:47: must set CVA6_REPO_DIR to point at the root of CVA6 sources -- doing it for you...
Makefile:144: XCELIUM_HOME not set which is necessary for compiling DPIs when using XCELIUM
rm -rf tmp/riscv-torture/output/test*
rm -rf work/ work-dpi/ work-ver/ work-vcs/
rm -f tmp/*.ucdb tmp/*.log *.wlf *vstf wlft* *.ucdb
cd corev_apu/fpga && make clean && cd ../..
make[1]: entering directory '/home/username/Documentos/cva6Repo/corev_apu/fpga'
rm -rf *.log *.jou *.str *.mif *.xpr work-fpga ariane.cache ariane.hw ariane.ip_user_files scripts/vivado*
make[1]: leaving directory '/home/username/Documentos/cva6Repo/corev_apu/fpga'
make: leaving directory '/home/username/Documentos/cva6Repo'
Makefile:14: must set CVA6_REPO_DIR to point at the root of CVA6 sources and CVA6_TB_DIR to point here -- doing it for you...
basename: one operand missing
Try 'basename --help' for more information.
[VCS] Cleanup (entire vcs_work dir)
rm -rf /home/username/Documentos/cva6Repo/verif/sim/vcs_results/ verdiLog/ simv* *.daidir *.vpd *.fsdb *.db csrc ucli.key vc_hdrs.h novas* inter.fsdb uart
rm -f *.txt
rm -f trace*.log
rm -f trace*.dasm
rm -f *.vpd *.fsdb *.vcd *.fst
/home/username/Documentos/cva6Repo
```
</details>

mkexc, 2023-11-17:
Hi @Juan-Gg, 

from your output log it appears that Verilator is not compiling correctly from source because your system is lacking the `help2man` package.

Try to install it, either by apt or building from source, and try relaunching the tests. I suggest you to start with the dhrystone test, then move to the smoke tests.

Juan-Gg, 2023-11-17:
Thank you, I did not see that.

I installed help2man and ran:
$ source verif/regress/dhrystone.sh

On the first run it seems to have compiled Verilator and it ended with this error:
```
ERROR    ERROR return code: True/2, cmd:make veri-testharness [...]
```

I ran it a second time, whose full log is below. It ended with the same error. Reding the output I noticed these lines but I don't know if it is the problem.

```
Makefile:14: must set CVA6_REPO_DIR to point at the root of CVA6 sources and CVA6_TB_DIR to point here -- doing it for you...
basename: missing an operand
Try 'basename --help' for more information.
```

<details>
  <summary><i>Output:</i></summary>

```
username:cva6Repo$ source verif/regress/dhrystone.sh 
+ source verif/regress/dhrystone.sh
++ '[' -n /opt/riscv ']'
++ source verif/regress/install-cva6.sh
+++++ dirname verif/regress/install-cva6.sh
++++ readlink -f verif/regress/../../
+++ export ROOT_PROJECT=/home/username/Documentos/cva6Repo
+++ ROOT_PROJECT=/home/username/Documentos/cva6Repo
+++ export TOP=/home/username/Documentos/cva6Repo/tools
+++ TOP=/home/username/Documentos/cva6Repo/tools
+++ '[' -z /opt/riscv ']'
+++ '[' -z riscv64-unknown-elf- ']'
+++ '[' -z riscv64-unknown-elf-gcc ']'
+++ '[' -z /opt/riscv/bin/riscv64-unknown-elf-objcopy ']'
+++ export PATH=/opt/riscv/bin:/home/username/Documentos/cva6Repo/tools/verilator-v5.008/bin:/home/username/Documentos/cva6Repo/tools/verilator-v5.008/bin:/opt/riscv/bin:/tools/Xilinx/Vitis_HLS/2020.2/bin:/tools/Xilinx/Vivado/2020.2/bin:/tools/Xilinx/Model_Composer/2020.2/bin:/tools/Xilinx/Vitis/2020.2/bin:/tools/Xilinx/Vitis/2020.2/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2020.2/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2020.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/Vitis/2020.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2020.2/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vitis/2020.2/aietools/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/tools:/opt/riscv/bin
+++ PATH=/opt/riscv/bin:/home/username/Documentos/cva6Repo/tools/verilator-v5.008/bin:/home/username/Documentos/cva6Repo/tools/verilator-v5.008/bin:/opt/riscv/bin:/tools/Xilinx/Vitis_HLS/2020.2/bin:/tools/Xilinx/Vivado/2020.2/bin:/tools/Xilinx/Model_Composer/2020.2/bin:/tools/Xilinx/Vitis/2020.2/bin:/tools/Xilinx/Vitis/2020.2/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2020.2/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2020.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/Vitis/2020.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2020.2/tps/lnx64/cmake-3.3.2/bin:/tools/Xilinx/Vitis/2020.2/aietools/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/tools:/opt/riscv/bin
+++ export LIBRARY_PATH=/opt/riscv/lib
+++ LIBRARY_PATH=/opt/riscv/lib
+++ export LD_LIBRARY_PATH=/opt/riscv/lib:/opt/riscv/lib:
+++ LD_LIBRARY_PATH=/opt/riscv/lib:/opt/riscv/lib:
+++ export C_INCLUDE_PATH=/opt/riscv/include
+++ C_INCLUDE_PATH=/opt/riscv/include
+++ export CPLUS_INCLUDE_PATH=/opt/riscv/include
+++ CPLUS_INCLUDE_PATH=/opt/riscv/include
+++ '[' /home/username/Documentos/cva6Repo/tools/verilator-v5.008 '!=' NO ']'
+++ source verif/regress/install-verilator.sh
++++ '[' -z 24 ']'
++++ '[' -z /home/username/Documentos/cva6Repo/tools ']'
++++ VERILATOR_REPO=https://github.com/verilator/verilator.git
++++ VERILATOR_BRANCH=master
++++ VERILATOR_HASH=v5.008
++++ VERILATOR_PATCH=/home/username/Documentos/cva6Repo/tools/../verif/regress/verilator-v5.patch
++++ '[' -n '' ']'
++++ '[' -z /home/username/Documentos/cva6Repo/tools/verilator-v5.008/verilator ']'
++++ '[' -z /home/username/Documentos/cva6Repo/tools/verilator-v5.008 ']'
++++ '[' '!' -f /home/username/Documentos/cva6Repo/tools/verilator-v5.008/bin/verilator ']'
++++ echo 'Using Verilator from cached directory /home/username/Documentos/cva6Repo/tools/verilator-v5.008.'
Using Verilator from cached directory /home/username/Documentos/cva6Repo/tools/verilator-v5.008.
++++ export PATH=/home/username/Documentos/cva6Repo/tools/verilator-v5.008/bin:/opt/riscv/bin:/home/username/Documentos/cva6Repo/tools/verilator-v5.008/bin:/home/username/Documentos/cva6Repo/tools/verilator-v5.008/bin:/opt/riscv/bin:/tools/Xilinx/Vitis_HLS/2020.2/bin:/tools/Xilinx/Vivado/2020.2/bin:/tools/Xilinx/Model_Composer/2020.2/bin:/tools/Xilinx/Vitis/2020.2/bin:/tools/Xilinx/Vitis/2020.2/gnu/microblaze/lin/bin:/tools/Xilinx/Vitis/2020.2/gnu/arm/lin/bin:/tools/Xilinx/Vitis/2020.2/gnu/microblaze/linux_toolchain/lin64_le/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-linux/bin:/tools/Xilinx/Vitis/2020.2/gnu/aarch64/lin/aarch64-none/bin:/tools/Xilinx/Vitis/2020.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/tools/Xilinx/Vitis/2020.2/tps/lnx64/cmake-3.3.2/ [... cut ...]
Fri, 17 Nov 2023 13:34:52 ERROR    ERROR return code: True/2, cmd:make veri-testharness target=gen64 variant=rv64imafdc_zicsr_zifencei elf=/home/username/Documentos/cva6Repo/verif/sim/out_2023-11-17/directed_c_tests/dhrystone_main.o path_var=/home/username/Documentos/cva6Repo/ tool_path=/home/username/Documentos/cva6Repo/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/username/Documentos/cva6Repo/verif/sim/out_2023-11-17/veri-testharness_sim/dhrystone_main.log &> /home/username/Documentos/cva6Repo/verif/sim/out_2023-11-17/veri-testharness_sim/dhrystone_main.log.iss

```

mkexc, 2023-11-17:
Could you please post also the content of the `/home/username/Documentos/cva6Repo/verif/sim/out_2023-11-17/veri-testharness_sim/dhrystone_main.log.iss` file?

Juan-Gg, 2023-11-17:
> Could you please post also the content of the `/home/username/Documentos/cva6Repo/verif/sim/out_2023-11-17/veri-testharness_sim/dhrystone_main.log.iss` file?

Added to the previous message.

Juan-Gg, 2023-11-30:
@valentinThomazic 
I built the toolchain as you suggested and it seems to be working now. Thank you.