openhwfoundation/cva6 #1638: Running hello.elf on Ariane 4.2 Simulation

hhhsiang, 2023-11-16:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I'm interested in conducting academic research on CPUs, and for this purpose, I've chosen Ariane (release 4.2) as it appears to be a stable version. After successfully running asm-tests and benchmarks using verilator4.110, everything seemed to be functioning well. However, when attempting to run a user application ('work-ver/Ariane_testharness -v hello.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf') following the README instructions, I encountered a simulation issue.
[image]
The simulation appears to be stuck, and there is no output or progress, even after waiting for two days. Additionally, I noticed that the hello.vcd file continuously grows and exceeds 200GB.

I came across similar issues reported in #282 and #298.

Since I need to run various test programs to validate my design, resolving this simulation problem for user applications is crucial.

I appreciate any assistance or insights you can provide. Thank you!

valentinThomazic, 2023-11-16:
Did you try to run your hello.elf using Spike and the pk ? Trying to do so may help knowing wether the issue is coming from your build of Ariane or your pk / hello.elf.
I suggest you use either a version of Spike you would build yourself or to use a version of this repo prior to https://github.com/openhwgroup/cva6/commit/220f534b6d30c9406ec2cccf5cb19d6a61c48895 since I personally did not manage to run the pk with Spike since these modifications even though there may be a way.

You can also try to run `$RISCV/riscv64-unknown-elf/bin/bbl` without your hello.elf and the -v option with Ariane. It is only the bootloader (so faster to load than the bbl, then the pk and finally your hello.elf) and may help you seeing if the simulation eventually finishes

JeanRochCoulon, 2023-11-20:
related to #1569

hhhsiang, 2023-11-24:
@JeanRochCoulon ,I ran user application(hello.elf) on ariane4v2(release4.2). I think that using verilator is the right way to run simulation for release4.2, and it is a different issue with #1569, right?

Moschn, 2023-11-24:
@hhhsiang release 4.2 is very old and does not really reflect the current state of CVA6. Since we work on master, it will be hard to help you with specific issues. Nevertheless, I can give you some pointers on how you could investigate your issues and fix them yourself.

From your description, I understand that simulation is actually already running; it just never terminates. Further steps now would involve figuring out why the hello world application does not terminate. As you already enabled trace generation, you can check which instructions are retired in the trace. The commit stage has two signals `commit_instr_i[0].pc` and `commit_instr_i[1].pc`, which contain the instructions that the core retires. In your case, you can use them to understand which instructions are executed/retired. This should help pin down the issues. 

Many people have been able to run a simple hello world at the time of release 4.2, so I assume the problems do not lie with the core nor the simulation (as you use the correct verilator version). Instead, I think the issues come from your toolchain and pk version. I have not yet seen any newish version of the pk or the toolchain be incompatible with the old release, but this could very well be the case.

hhhsiang, 2023-11-25:
@Moschn ,thanks for your help! I changed to older version of pk, and it finally worked!

valentinThomazic, 2023-11-30:
Hey @hhhsiang, could you close the issue if your problem is solved ?

yulong-lan, 2024-03-09:
Hi @hhhsiang. I run into the same problem as you did and I am also using an older version of everything. Could you tell me how you changed pk version and which, to run an elf?

hhhsiang, 2024-03-11:
Hi @yulong-lan. You can try to change the version of riscv-pk to 46cd5082c5d324bb843be35d8130aa9d44068d7d. You can go to newer version of cva6 to copy .sh file to intall riscv-pk. I go to newer commit  (about 2023/3), and copy the install-riscvpk.sh file then change the pk version in it.
[image]

yulong-lan, 2024-03-14:
@hhhsiang  have you tried to run an elf program directly without using proxy kernel?

hhhsiang, 2024-03-15:
@yulong-lan if you want to print "Hello CVA6", you can search related issue, some people have proposed it. You just need to modify some code, and it will print mormally.

> @hhhsiang have you tried to run an elf program directly without using proxy kernel?

I didn't try it ever, but i think i will need to run elf program in the future. Maybe we can discuss together.