openhwfoundation/cva6 #1643: [BUG] Issue compiling the corev_apu

cfuguet, 2023-11-19:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello everyone,

Is someone else having issues to compile the corev_apu/tb testbench in the CVA6 with Verilator ?

I'm having the following error: read_elf, get_section, and read_section_void symbols undefined.

It looks like the call to these functions were introduced by commit 220f534b6d30c9406ec2cccf5cb19d6a61c48895 but I don't see where there are defined.

Thanks for the help :)

JeanRochCoulon, 2023-11-20:
This command "make verilate" is not supported. This command implies the setup of other variables to be functional. And we are about to suppress such commands from the README.md. You have to use the cva6.py as simulation entry point.

JeanRochCoulon, 2023-11-20:
related to #1569

valentinThomazic, 2023-12-01:
Hi, did you manage to solve your issue ?