openhwfoundation/cva6 #1691: [TASK] generate documentation (User Manual and Design Documentation) with templating

JeanRochCoulon, 2023-12-10:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Several cva6 configurations must co-exit in documentations
Templating will allow to maintain one single master document and generate all the configuration documents 

### Required Changes

Create a master document with templating. All configurations will be specified in it. Then different configuration document will be generated from it.

### Current Status

All configurations are described in User Manual.
Design document describes only one configuration: the embedded configuration.

### Risks

Warning: the templating should be easy to maintain to allow everybody to write specifications

### Prerequisites

none

### KPI (KEY Performance Indicators)

none

### Description of Done

Commit of User Manual and Design Document master doc

JeanRochCoulon, 2023-12-10:
@jquevremont

jquevremont, 2023-12-11:
Hi @JeanRochCoulon, I have completed the description so that the solution is reviewed before deployment.