openhwfoundation/cva6 #1693: [BUG]  Instruction are incomplete

rajatkmitra, 2023-12-11:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Seems like the cva6 home page was updates (removed support for pk). However in scetion (2) "Install the GCC Toolchain" the instructions got removed. Please replace the instructions as I am rebuilding the cva6 area locally and am stuck w/o this.

cathales, 2023-12-11:
Is it what you are looking for? https://github.com/openhwgroup/cva6/tree/master/util/gcc-toolchain-builder

rajatkmitra, 2023-12-11:
Hello,
 Yes, this looks correct. I am building the environment. I need to create tool 64bit toolchain for bare metal implementation. 
I ab building using -
%sh get-toolchain.sh gcc-10.2.0-baremetal
%sh build-toolchain.sh gcc-13.1.0-baremetal $RISCV
Would these create 64bit version of the toolchain ?
Thanks,
Raj

rajatkmitra, 2023-12-11:
While compiling the toolchain, I am encountering this error -
In file included from ../../../../../../src/gcc/libgcc/libgcov-merge.c:26:
../../../../../../src/gcc/libgcc/libgcov.h:49:10: fatal error: sys/mman.h: No such file or directory
   49 | #include <sys/mman.h>
      |          ^~~~~~~~~~~~
compilation terminated.

It seems also that the toolchain is getting build for 32bit implementation, how will I force script to create 64bit toolchain ?
Thanks,
Raj

rajatkmitra, 2023-12-15:
Any updates for me ??

valentinThomazic, 2023-12-27:
There was a broken link to the toolchain build script README. 
It is now fixed.
Did you install all the build prerequisites ? 
The built toolchain would indeed default to 32 bits if used standalone but you can use the `cv64a6_imafdc_sv39` target with `cva6.py` (see the `smoke-test.sh`).

rajatkmitra, 2024-01-02:
Okay thanks, I will giv ethis a try and let you know shortly..
--Raj

rajatkmitra, 2024-01-04:
Still environment is brokev, I did a fresh checkout and then 
pip3 install -r verif/sim/dv/requirements.txt
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'verif/sim/dv/requirements.txt'

rajatkmitra, 2024-01-04:
All build prerequisites are now installed, however in the latest cva6 download, with the following instructions -
========================================================================
Once the prerequisites (see above) are satisfied, you can fetch and build the upstream GCC toolchain (default: 13.1.0) for bare-metal 32-bit and 64-bit applications in just three steps.

# 1. Select an installation location for the toolchain (here: the default RISC-V tooling directory $RISCV).
INSTALL_DIR=$RISCV

# 2. Fetch the source code of the toolchain (assumes Internet access.)
sh get-toolchain.sh

# 3. Build and install the toolchain (requires write+create permissions for $INSTALL_DIR.)
You can set 
sh build-toolchain.sh $INSTALL_DIR
========================================================================

You would not  find get-toolchain.sh,  build-toolchain.sh...
Please add locations  for these - cva6/util/gcc-toolchain-builder... in your instructions... 

Are you suggesting that if I build bare metal  -
sh build-toolchain.sh -f gcc-13.1.0-baremetal $RISCV

then I could still run the test for 64 bit with cva6.py and choosing the cv64a6_imafdc_sv39  target ?? Also when is the smoke-test.sh located ??

rajatkmitra, 2024-01-05:
The problem still exists in building the toochain -

% sh build-toolchain.sh gcc-13.1.0-baremetal $RISCV

In file included from ../../../../../../src/gcc/libgcc/libgcov-merge.c:26:
../../../../../../src/gcc/libgcc/libgcov.h:49:10: fatal error: sys/mman.h: No such file or directory
   49 | #include <sys/mman.h>
      |          ^~~~~~~~~~~~
compilation terminated.
Makefile:924: recipe for target '_gcov_merge_add.o' failed
make[4]: *** [_gcov_merge_add.o] Error 1
make[4]: Leaving directory '/home/rajatkmitra/risc5/pulp/cva6/1.2024/cva6/util/gcc-toolchain-builder/build/gcc/riscv-none-elf/rv32i/ilp32/libgcc'
Makefile:1214: recipe for target 'multi-do' failed
make[3]: *** [multi-do] Error 1
make[3]: Leaving directory '/home/rajatkmitra/risc5/pulp/cva6/1.2024/cva6/util/gcc-toolchain-builder/build/gcc/riscv-none-elf/libgcc'
Makefile:125: recipe for target 'all-multi' failed
make[2]: *** [all-multi] Error 2
make[2]: Leaving directory '/home/rajatkmitra/risc5/pulp/cva6/1.2024/cva6/util/gcc-toolchain-builder/build/gcc/riscv-none-elf/libgcc'
Makefile:13205: recipe for target 'all-target-libgcc' failed
make[1]: *** [all-target-libgcc] Error 2
make[1]: Leaving directory '/home/rajatkmitra/risc5/pulp/cva6/1.2024/cva6/util/gcc-toolchain-builder/build/gcc'
Makefile:1018: recipe for target 'all' failed
make: *** [all] Error 2
*** Could not build GCC (even after removing target dirs), bailing out!

valentinThomazic, 2024-01-05:
> Are you suggesting that if I build bare metal -
sh build-toolchain.sh -f gcc-13.1.0-baremetal $RISCV
then I could still run the test for 64 bit with cva6.py and choosing the cv64a6_imafdc_sv39 target ?? Also when is the smoke-test.sh located ??

Yes indeed. I am not sure about your second question but the smoke tests are in `verif/regress/smoke-tests.sh` if this is what you asked.

About your toolchain build error, you are probably lacking a build prerequisite package. The build prerequisites given in the toolchain build scripts README are the one given in the [official riscv gcc toolchain repo](https://github.com/riscv-collab/riscv-gnu-toolchain) but it is not said they are exhaustive. 
I see you use a rather old version of Ubuntu, this could be part of the issue. A quick google search lead me to think you should try to install or reinstall `libc6-dev` but I cannot guarantee this will suffice or help at all.

rajatkmitra, 2024-01-05:
Thanks, I ran on the latest Ubuntu release and the tool chain compiled  and the smoke tests appear to have run. Thanks for your support. Is it possible for me to write tests and run them individually and look a waveforms (VCD files) using GTKWave. If so how might this be done ? Are there any instructions ??

rajatkmitra, 2024-01-05:
I tried running the hello world test-
=====================================================================
# Make sure to source this script from the root directory 
# to correctly set the environment variables related to the tools
source verif/sim/setup-env.sh

# Set the NUM_JOBS variable to increase the number of parallel make jobs
# export NUM_JOBS=

export DV_SIMULATORS=veri-testharness

cd ./verif/sim

python3 cva6.py --target cv32a60x --iss=$DV_SIMULATORS --iss_yaml=cva6.yaml \
--c_tests ../tests/custom/hello_world/hello_world.c \
--linker=../tests/custom/common/test.ld \
--gcc_opts="-static -mcmodel=medany -fvisibility=hidden -nostdlib \
-nostartfiles -g ../tests/custom/common/syscalls.c \
../tests/custom/common/crt.S -lgcc \
-I../tests/custom/env -I../tests/custom/common"
=====================================================================
But I am erroring out on some Ariane  testbench code -

valentinThomazic, 2024-01-05:
I tried running the commands from a blank repo and did not encounter this issue.
Can you run the smoke-tests in this state?

rajatkmitra, 2024-01-05:
Thanks for the pointer, I started running the smoke tests and looked closely, I see script trying to build spike in root folder -

Installing Spike in '/home/rajatkmitra/cva6/cva6/tools/spike'...
../scripts/mk-install-dirs.sh /include
mkdir /include
mkdir: cannot create directory ‘/include’: Permission denied
make: *** [Makefile:395: install-hdrs] Error 1
Repo:   https://github.com/riscv-non-isa/riscv-arch-test.git
Branch: main
Hash:   220e78542da4510e40eac31e31fdd4e77cdae437
Patch:  ../../../verif/regress/riscv-compliance.patch
Repo:   https://github.com/riscv/riscv-tests.git
Branch: master
Hash:   f92842f91644092960ac7946a61ec2895e543cec
Building Spike in /home/rajatkmitra/cva6/cva6/verif/core-v-verif/vendor/riscv/riscv-isa-sim
Installing Spike in /home/rajatkmitra/cva6/cva6/tools/spike
Building Spike sources in /home/rajatkmitra/cva6/cva6/verif/core-v-verif/vendor/riscv/riscv-isa-sim...
make: Circular libfesvr.so <- libfesvr.so dependency dropped.
make: Circular libriscv.so <- libriscv.so dependency dropped.
make: Circular libdisasm.so <- libdisasm.so dependency dropped.
make: Circular libcustomext.so <- libcustomext.so dependency dropped.
make: Circular libsoftfloat.so <- libsoftfloat.so dependency dropped.
make: Nothing to be done for 'default'.
Installing Spike in '/home/rajatkmitra/cva6/cva6/tools/spike'...
../scripts/mk-install-dirs.sh /include
mkdir /include
mkdir: cannot create directory ‘/include’: Permission denied
make: *** [Makefile:395: install-hdrs] Error 1
Repo:   https://github.com/riscv-non-isa/riscv-arch-test
Branch: main
Hash:   a5a49fc9f244192649e57fe61b4513d9bc39b1e3
Fri, 05 Jan 2024 11:40:33 INFO     GCC Version : 13.1.0
Fri, 05 Jan 2024 11:40:33 INFO     Spike Version : /home/rajatkmitra/cva6/cva6/tools/spike
Fri, 05 Jan 2024 11:40:33 INFO     Verilator Version : Verilator 5.018 2023-10-30 rev v5.018

I do not have permission to create spike in root directory

rajatkmitra, 2024-01-05:
Following that there are other errors.. 
Fri, 05 Jan 2024 11:40:33 INFO     GCC Version : 13.1.0
Fri, 05 Jan 2024 11:40:33 INFO     Spike Version : /home/rajatkmitra/cva6/cva6/tools/spike
Fri, 05 Jan 2024 11:40:33 INFO     Verilator Version : Verilator 5.018 2023-10-30 rev v5.018

Fri, 05 Jan 2024 11:40:33 INFO     Creating output directory: /home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05

Fri, 05 Jan 2024 11:40:33 INFO     Execution numero : 1
Fri, 05 Jan 2024 11:40:33 INFO     Processing regression test list : ../tests/testlist_riscv-tests-cv64a6_imafdc_sv39-v.yaml, test: rv64ui-v-add
Fri, 05 Jan 2024 11:40:33 INFO     Found matched tests: rv64ui-v-add, iterations:1
Fri, 05 Jan 2024 11:40:33 INFO     CVA6 Configuration is 
Fri, 05 Jan 2024 11:40:33 INFO     Compiling assembly test : /home/rajatkmitra/cva6/cva6/verif/tests/riscv-tests/isa/rv64ui/add.S
/home/rajatkmitra/cva6/toolchain/lib/gcc/riscv-none-elf/13.1.0/../../../../riscv-none-elf/bin/ld: warning: /home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/directed_asm_tests/add.o has a LOAD segment with RWX permissions
Fri, 05 Jan 2024 11:40:33 INFO     Converting to /home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/directed_asm_tests/add.bin
Fri, 05 Jan 2024 11:40:33 INFO     Processing ISS setup file : cva6.yaml
Fri, 05 Jan 2024 11:40:34 INFO     Found matching ISS: veri-testharness
Fri, 05 Jan 2024 11:40:34 INFO     ISA rv64gc_zba_zbb_zbs_zbc_zicsr_zifencei
Fri, 05 Jan 2024 11:40:34 INFO     [veri-testharness] Running ISS simulation: make veri-testharness target=cv64a6_imafdc_sv39 variant=rv64gc_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/directed_asm_tests/add.o path_var=/home/rajatkmitra/cva6/cva6/ tool_path=/home/rajatkmitra/cva6/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/veri-testharness_sim/add.log &> /home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/veri-testharness_sim/add.log.iss
Fri, 05 Jan 2024 11:40:35 INFO     
Fri, 05 Jan 2024 11:40:35 ERROR    ERROR return code: True/2, cmd:make veri-testharness target=cv64a6_imafdc_sv39 variant=rv64gc_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/directed_asm_tests/add.o path_var=/home/rajatkmitra/cva6/cva6/ tool_path=/home/rajatkmitra/cva6/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/veri-testharness_sim/add.log &> /home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/veri-testharness_sim/add.log.iss
Fri, 05 Jan 2024 11:40:35 INFO     GCC Version : 13.1.0
Fri, 05 Jan 2024 11:40:35 INFO     Spike Version : /home/rajatkmitra/cva6/cva6/tools/spike
Fri, 05 Jan 2024 11:40:35 INFO     Verilator Version : Verilator 5.018 2023-10-30 rev v5.018

Fri, 05 Jan 2024 11:40:35 INFO     Creating output directory: /home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05

Fri, 05 Jan 2024 11:40:35 INFO     Execution numero : 1
Fri, 05 Jan 2024 11:40:35 INFO     Processing regression test list : ../tests/testlist_riscv-tests-cv64a6_imafdc_sv39-p.yaml, test: rv64ui-p-add
Fri, 05 Jan 2024 11:40:35 INFO     Found matched tests: rv64ui-p-add, iterations:1
Fri, 05 Jan 2024 11:40:35 INFO     CVA6 Configuration is 
Fri, 05 Jan 2024 11:40:35 INFO     Compiling assembly test : /home/rajatkmitra/cva6/cva6/verif/tests/riscv-tests/isa/rv64ui/add.S
Fri, 05 Jan 2024 11:40:35 INFO     Converting to /home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/directed_asm_tests/add.bin
Fri, 05 Jan 2024 11:40:35 INFO     Processing ISS setup file : cva6.yaml
Fri, 05 Jan 2024 11:40:35 INFO     Found matching ISS: veri-testharness
Fri, 05 Jan 2024 11:40:35 INFO     ISA rv64gc_zba_zbb_zbs_zbc_zicsr_zifencei
Fri, 05 Jan 2024 11:40:35 INFO     [veri-testharness] Running ISS simulation: make veri-testharness target=cv64a6_imafdc_sv39 variant=rv64gc_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/directed_asm_tests/add.o path_var=/home/rajatkmitra/cva6/cva6/ tool_path=/home/rajatkmitra/cva6/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/veri-testharness_sim/add.log &> /home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/veri-testharness_sim/add.log.iss
Fri, 05 Jan 2024 11:40:36 INFO     
Fri, 05 Jan 2024 11:40:36 ERROR    ERROR return code: True/2, cmd:make veri-testharness target=cv64a6_imafdc_sv39 variant=rv64gc_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/directed_asm_tests/add.o path_var=/home/rajatkmitra/cva6/cva6/ tool_path=/home/rajatkmitra/cva6/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/veri-testharness_sim/add.log &> /home/rajatkmitra/cva6/cva6/verif/sim/out_2024-01-05/veri-testharness_sim/add.log.iss
Fri, 05 Jan 2024 11:40:36 INFO     GCC Version : 13.1.0
Fri, 05 Jan 2024 11:40:36 INFO     Spike Version : /home/rajatkmitra/cva6/cva6/tools/spike
Fri, 05 Jan 2024 11:40:36 INFO     Verilator Version : Verilator 5.018 2023-10-30 rev v5.018

[... cut ...]
make: Leaving directory '/home/rajatkmitra/cva6/cva6'
Makefile:14: must set CVA6_REPO_DIR to point at the root of CVA6 sources and CVA6_TB_DIR to point here -- doing it for you...
basename: missing operand
Try 'basename --help' for more information.
[VCS] Cleanup (entire vcs_work dir)
rm -rf /home/rajatkmitra/cva6/cva6/verif/sim/vcs_results/ verdiLog/ simv* *.daidir *.vpd *.fsdb *.db csrc ucli.key vc_hdrs.h novas* inter.fsdb uart
rm -f *.txt
rm -f trace*.log
rm -f trace*.dasm
rm -f *.vpd *.fsdb *.vcd *.fst
/home/rajatkmitra/cva6/cva6

rajatkmitra, 2024-01-05:
The smoke tests ended and I grepped teh failures - 

base) rajatkmitra@rajatkmitra-ThinkCentre-M93p:~/cva6/cva6/verif/sim/out_2024-01-05/veri-testharness_sim$ grep "No such file or directory" *
add-01.log.iss:../corev_apu/tb/dpi/SimDTM.cc:4:10: fatal error: fesvr/dtm.h: No such file or directory
add-01.log.iss:../corev_apu/tb/ariane_tb.cpp:42:10: fatal error: fesvr/dtm.h: No such file or directory
add.log.iss:../corev_apu/tb/dpi/SimDTM.cc:4:10: fatal error: fesvr/dtm.h: No such file or directory
add.log.iss:../corev_apu/tb/ariane_tb.cpp:42:10: fatal error: fesvr/dtm.h: No such file or directory
cadd-01.log.iss:../corev_apu/tb/dpi/SimDTM.cc:4:10: fatal error: fesvr/dtm.h: No such file or directory
cadd-01.log.iss:../corev_apu/tb/ariane_tb.cpp:42:10: fatal error: fesvr/dtm.h: No such file or directory
custom_test_template.log.iss:../corev_apu/tb/dpi/SimDTM.cc:4:10: fatal error: fesvr/dtm.h: No such file or directory
custom_test_template.log.iss:../corev_apu/tb/ariane_tb.cpp:42:10: fatal error: fesvr/dtm.h: No such file or directory
hello_world.log.iss:../corev_apu/tb/dpi/SimDTM.cc:4:10: fatal error: fesvr/dtm.h: No such file or directory
hello_world.log.iss:../corev_apu/tb/ariane_tb.cpp:42:10: fatal error: fesvr/dtm.h: No such file or directory
I-ADD-01.log.iss:../corev_apu/tb/dpi/SimDTM.cc:4:10: fatal error: fesvr/dtm.h: No such file or directory
I-ADD-01.log.iss:../corev_apu/tb/ariane_tb.cpp:42:10: fatal error: fesvr/dtm.h: No such file or directory

rajatkmitra, 2024-01-10:
Are there any suggestions / updates ??
--Raj

rajatkmitra, 2024-01-16:
It's been a while... Has this been fixed ??
--Raj

zchamski, 2024-02-09:
Hi @rajatkmitra  and @valentinThomazic , the invalid path `/include` and the missing file errors in logs above point to a missing/incomplete Spike installation.

A standard out-of-the-box build of the CVA support tools installs Spike in `<cva6_top_dir>/tools/spike`.  During script execution, the corresponding absolute path is stored in shell variable `SPIKE_INSTALL_DIR`.

@rajatkmitra, can you please set `SPIKE_INSTALL_DIR` as described just above, run the smoke tests and attach the log (preferably as a file, not pasted text)?

valentinThomazic, 2024-02-21:
Hey @rajatkmitra, any news ?

rajatkmitra, 2024-02-21:
Hi @valentinThomazic and @zchamski 
Apologies !! I was stuck with upgrading Ubuntu O/S and have not tried yet. Please allow me a few more days. I will install Spike seperately and create the SPIKE_INSTALL_DIR env variable and reinstall cva6 . Then I will try running the spoke test and update you with my finding. Will update early next week ..
Many thanks for your continued support !!!
--Raj

valentinThomazic, 2024-02-21:
No worries ! I made some changes in the installation scripts, I recommend you try to pull the last version of the repo and run the smoke-tests again before bothering installing Spike manually.

rajatkmitra, 2024-02-21:
Great,
Will do that and inform shortly ;-)
--R

valentinThomazic, 2024-03-29:
Hey @rajatkmitra, it's been a month. Did you manage to setup the repo ?

rajatkmitra, 2024-03-29:
Hi @valentinThomazic 
Many apologies, I was out sick for a while ;-(... just catching up .. I will try this weekend and update.
Thanks for your understanding,
Raj

rajatkmitra, 2024-03-29:
Hi @valentinThomazic 
All tests have passed !! Thanks fo ryour help with this. Can you explain how to generated VCD Files to observe test waveforms ??
Thanks,
Raj

rajatkmitra, 2024-03-29:
Hi @valentinThomazic 
I just ran with waveform dump enabled, -
export DV_SIMULATORS=veri-testharness,spike
export TRACE_FAST=1
bash verif/regress/smoke-tests.sh
but cannot locate the  VCD files. Where should they be ??

valentinThomazic, 2024-04-02:
Hey @rajatkmitra ! That's great ! 
About the description of the tests, The smoke tests script runs test from the different testsuites referenced in the README. I suggest you either look at the test suites description and/or tests source files or run your own program so you're sure to know exactly what it does.

cathales, 2024-04-02:
For the VCD file, IIRC it should be in `verif/sim/out_$(date -I)/veri-testharness_sim/`

JeanRochCoulon, 2024-04-03:
I think this github issue can be closed ! @rajatkmitra can you ?

rajatkmitra, 2024-04-03:
Ye I will close it.. I will file anothere issue for the test description  if I need more help after folloing @valentinThomazic  instructions. Alos will look for VCD files as suggested by @cathales 
Many thanks for your support !

closed 2024-04-03 by rajatkmitra