openhwfoundation/cva6 #1700: [BUG] [ERROR return code: True/2, cmd:make veri-testharness] when running simulations

Siris-Limx, 2023-12-12:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello!
I'm new to deploy CVA6 and met with some problems when I simulated the CVA6 by using `verif/sim/cva6.py`.
I carefully followed the `README.md` and successfully installed the GCC toolchain and Verilator.
```
$ riscv-none-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv-none-elf-gcc
COLLECT_LTO_WRAPPER=/home/sirisli/Projects/cva6/tools/riscv_toolchain/libexec/gcc/riscv-none-elf/13.1.0/lto-wrapper
Target: riscv-none-elf
Configured with: ../../src/gcc/configure --prefix=/home/sirisli/Projects/cva6/tools/riscv_toolchain --target=riscv-none-elf --enable-languages=c --disable-libssp --disable-libgomp --disable-libmudflap
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.1.0 (GCC) 

$ verilator -V
Verilator 5.008 2023-03-04 rev v5.008 (mod)

Copyright 2003-2023 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.
```

However, when I tried to simulate the CVA6 by running either `verif/regress/*.sh` or `verif/sim/cva6.py` directly, I will get such error:
```
Tue, 12 Dec 2023 19:13:52 ERROR    ERROR return code: True/2, cmd:make veri-testharness target=cv32a60x variant=rv32imac_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/sirisli/Projects/cva6/verif/sim/out_2023-12-12/directed_c_tests/hello_world.o path_var=/home/sirisli/Projects/cva6/ tool_path=/home/sirisli/Projects/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+debug_disable=1 +ntb_random_seed=1" isspostrun_opts="0x0000000080000000" log=/home/sirisli/Projects/cva6/verif/sim/out_2023-12-12/veri-testharness_sim/hello_world.log &> /home/sirisli/Projects/cva6/verif/sim/out_2023-12-12/veri-testharness_sim/hello_world.log.iss
```

I've also referred to the log file:
```
make[2]: Entering directory '/home/sirisli/Projects/cva6/work-ver'
g++    ariane_tb.o SimDTM.o SimJTAG.o msim_helper.o remote_bitbang.o verilated.o verilated_dpi.o verilated_vpi.o verilated_threads.o Variane_testharness__ALL.a   -L/home/sirisli/Projects/cva6/tools/riscv_toolchain/lib -L/home/sirisli/Projects/cva6/tools/spike/lib -Wl,-rpath,/home/sirisli/Projects/cva6/tools/riscv_toolchain/lib -Wl,-rpath,/home/sirisli/Projects/cva6/tools/spike/lib -lfesvr -lriscv  -lpthread  -pthread -lpthread -latomic   -o Variane_testharness
/usr/bin/ld: ariane_tb.o: in function `main':
ariane_tb.cpp:(.text.startup+0x64a): undefined reference to `read_elf'
/usr/bin/ld: ariane_tb.cpp:(.text.startup+0x6a8): undefined reference to `get_section'
/usr/bin/ld: ariane_tb.cpp:(.text.startup+0x6db): undefined reference to `read_section_void'
collect2: error: ld returned 1 exit status
make[2]: *** [Variane_testharness.mk:80: Variane_testharness] Error 1
make[2]: Leaving directory '/home/sirisli/Projects/cva6/work-ver'
make[1]: *** [Makefile:583: verilate] Error 2
make[1]: Leaving directory '/home/sirisli/Projects/cva6'
make: *** [Makefile:140: veri-testharness] Error 2
```

Siris-Limx, 2023-12-13:
I've found one previous issue #1631 that encountered the same problems as mine.

JeanRochCoulon, 2023-12-13:
Thanks for the information. @valentinThomazic will maybe add this information to the README.md.

valentinThomazic, 2023-12-27:
It has been added to the README, thank you @Siris-Limx for pointing that out.
If you are ok with the changes, please close your issue.

Siris-Limx, 2023-12-27:
> If you are ok with the changes, please close your issue.

Thanks for your changes, I close this issue right now.

Siris-Limx, 2024-01-11:
Hello! I've found a bug in the first run of `bash verif/regress/smoke-tests.sh` to install spike.

Here is the code segment of `verif/regress/install-spike.sh`:
```
  # Build and install Spike (including extensions).
  mkdir -p build
  cd build
  if [[ ! -f config.log ]]; then
      ../configure --prefix="$SPIKE_INSTALL_DIR"
  fi
  make -j${NUM_JOBS}
  echo "Installing Spike in '$SPIKE_INSTALL_DIR'..."
  make install
  cd $CALLER_DIR
```
Here, I think we should add `sudo` before `make install`,  that is `sudo make install` just as the `README` recommended in spike original repository, otherwise there may exists some `permission denied` error.

JeanRochCoulon, 2024-01-11:
@Siris-Limx    Your first github issue cannot be reused for another issue. That's why I close definitively this one and please open another one with your new issue. That's the way to use "github issue". Thank you

closed 2024-01-11 by JeanRochCoulon