openhwfoundation/cva6 #1715: [BUG] No documentation on CVA 6 Boot Process

rajatkmitra, 2023-12-18:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I need help understanding the CVA6 boot process -
I created a simple bare metal set of instructions -

Disassembly of section .text:

0000000000000000 <main>:
   0:   1101                    c.addi  x2,-32
   2:   ec22                    c.sdsp  x8,24(x2)
   4:   1000                    c.addi4spn      x8,x2,32
   6:   4785                    c.li    x15,1
   8:   fef42623                sw      x15,-20(x8)
   c:   4789                    c.li    x15,2
   e:   fef42423                sw      x15,-24(x8)
  12:   fec42783                lw      x15,-20(x8)
  16:   873e                    c.mv    x14,x15
  18:   fe842783                lw      x15,-24(x8)
  1c:   9fb9                    c.addw  x15,x14
  1e:   fef42223                sw      x15,-28(x8)
  22:   0001                    c.addi  x0,0
  24:   853e                    c.mv    x10,x15
  26:   6462                    c.ldsp  x8,24(x2)
  28:   6105                    c.addi16sp      x2,32
  2a:   8082                    c.jr    x1
and and am loading boot code through the AXI interface. It seems AXI interface reade the first several instructions and then stops 

[image]

1. Where are the registers in the device to look at the various register operations completing (verilog hierarchy) ?
2. Which signal in the verilog hierarchy do I look to find if exceptions are generated ?
3. Why does the AXI bus not complete reading after the first 4 DWords ?
Thanks,
Raj

JeanRochCoulon, 2023-12-27:
Hello @AEzzejjari   As you are familiar with AXI transactions, do you have some answers to @rajatkmitra questions ?

JeanRochCoulon, 2025-01-02:
@rajatkmitra We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it ?

rajatkmitra, 2025-01-02:
Please close