openhwfoundation/cva6 #1716: [Spike] : unsupported unprivileged counters CSRs : cycle[h], instret[h]

AyoubJalali, 2023-12-18:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Spike don't support unprivileged counters CSRs, as cycle[h], instret[h].
To reproduce the bug, just try to read from this CSR, and you gonna get an exception raised (don't try to write because it's RO csrs, so the right decision is to raise an exception)

here is an example of spike.log
[image]

MarioOpenHWGroup, 2024-01-05:
After reproducing the issue, I have realized that there is a missing parameter for the `zicntr` extension which is not by default active in Spike. In my opinion all the extensions should have a parameter in the cva6cfg.

MikeOpenHWGroup, 2024-01-05:
@MarioOpenHWGroup, are you saying:
- that Spike can be configured to support the unprivileged counters CSRs such as cycle[h], instret[h] when `zicntr` is enabled?
- we should add a new member to [uvme_cva6_cfg.sv](https://github.com/openhwgroup/cva6/blob/c430c6c34be0e24175e03d3c91478e349bfaaf4e/verif/env/uvme/uvme_cva6_cfg.sv#L29) to indicate that the CVA6 supports Zicntr?

If so, I assume either the default constraints in [uvme_cva6_cfg.sv](https://github.com/openhwgroup/cva6/blob/c430c6c34be0e24175e03d3c91478e349bfaaf4e/verif/env/uvme/uvme_cva6_cfg.sv#L88) or the UVM testcase constrains the value of that new member (say `ext_zicntr_supported`) to indicate that the Zicntr is supported.

If all of that is correct, how do we use `uvme_cva6_cfg_c::ext_zicntr_supported` to configure the Spike model.  Where is this code to configure Spike located?

MarioOpenHWGroup, 2024-01-05:
Yes Spike can be configured adding the `zicntr` to the ISA string. 
In my opinion as this is an extension should be treated as it is, yes.

I noticed that on Spike and I have some local changes pending to be pushed as a PR to fix this.

AyoubJalali, 2024-01-08:
I confirm that spike support unprivileged counters when **zicntr** is enabled

zchamski, 2024-04-04:
@AyoubJalali: since enabling the `Zicntr` ISA extension solves this issue, can you please close it?

closed 2024-04-04 by AyoubJalali