openhwfoundation/cva6 #1720: [TASK] Add a test for misaligned branches

cathales, 2023-12-19:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Issue #1300 should be fixed by PR #1719. There is no regression, but we lack a test to verify that the bug is actually fixed. (With TDD we would have added this test before fixing the bug)

### Required Changes

Add a test branching to a `target_address` such as `target_address[1:0]` is `2’b10`.

### Current Status

There is no test to verify that the #1300 is fixed.

### Description of Done

A misaligned address exception shall be raised if RVC extension is disabled, but not if RVC extension is enabled.

JeanRochCoulon, 2023-12-27:
@AyoubJalali  Can you tell me whether this test should be linked to ISACOV ?   Is it possible to generate it with riscv-dv ?

JeanRochCoulon, 2024-01-27:
@AyoubJalali   Any idea ?

AyoubJalali, 2024-06-03:
> @AyoubJalali Any idea ?

we can't raise this exception, because we support C extension, so it means that we should Execute on an address half-work misaligned LSB bit != 0 -> and this can't never happen in the RISC-V architecture because The jump and branch interuction always clear the LSB bit

JeanRochCoulon, 2024-06-03:
As it is not 65x, we keep the github issu eopen but I change the assignee

closed 2026-09-07 by JeanRochCoulon