openhwfoundation/cva6 #1726: [BUG] ASIC synthesis failed related to ariane_pkg.sv

andriami, 2023-12-19:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi, 

I am trying to implement the CVA6 on ASIC by using the proposed flow. I have errors related to ariane_pkg.sv synthesis.

    Compiling source file ../..//core/include/ariane_pkg.sv
    Error:  ../..//core/include/ariane_pkg.sv:169: Package 'ariane_pkg' has not been analyzed for import or content extraction. (VER-224) 

This is due to the use of ariane_pkg inside the ariane_pkg itself.

    localparam int unsigned INSTR_PER_FETCH = RVC == 1'b1 ? (FETCH_WIDTH / 16) : 1;
    localparam int unsigned LOG2_INSTR_PER_FETCH = RVC == 1'b1 ? $clog2(
       ariane_pkg::INSTR_PER_FETCH
    ) : 1;

To resolve this issue, I removed the ariane_pkg before INSTR_PER_FETCH.

    localparam int unsigned INSTR_PER_FETCH = RVC == 1'b1 ? (FETCH_WIDTH / 16) : 1;
    localparam int unsigned LOG2_INSTR_PER_FETCH = RVC == 1'b1 ? $clog2(
       INSTR_PER_FETCH
    ) : 1;

Do you know another way of solving this issue?

Thanks

JeanRochCoulon, 2023-12-21:
Hello @andriami 
The ASIC synthesis runs on CI. We run also regularly it in interactive way. I do not see why you get a bug due to RTL. What is your Dc_shell version ?

andriami, 2023-12-21:
Hi @JeanRochCoulon,

Thank you for your answer!
I am using version P-2019.03 of DC.

JeanRochCoulon, 2023-12-27:
We use S-2021.09-SP1-1. Thsi can be the root cause.
As your suggested fix seems good to me. Can I ask you to submit a PR ?

JeanRochCoulon, 2024-05-22:
CLosed because fixed by #1750.

closed 2024-05-22 by JeanRochCoulon

andriami, 2024-10-15:
Hi @hhhsiang, I successfully synthesized it using the May 22nd, 2024 version, but I haven't tried it with the latest version yet.

hhhsiang, 2024-10-15:
@andriami ,did you synthesize it with the makefile under `pd/synth`, or you create a script on your own? Which config pkg did you use to synthesize? Is that `cv64a6_imafdc_sv39_config_pkg.sv`?

I encounter error like this issue:https://github.com/openhwgroup/cva6/issues/2544
Thanks in advance!

andriami, 2024-10-28:
Hi @hhhsiang, yes I used the makefile under pd/syn and the cv64a6_imafdc_sv39_config_pkg.sv config. I didn't encounter the issue #2544 you mentioned. For information, we have used an older version ( Q-2019.12) of DC.