openhwfoundation/cva6 #1727: Do we still need install-cva6.sh?

MikeOpenHWGroup, 2023-12-19:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Now that the UVM environment has been integrated into this repo and [core-v-verif](https://github.com/openhwgroup/core-v-verif) is a submodule of this repo it is not clear that the purpose of the [install-cva6.sh](https://github.com/openhwgroup/cva6/blob/2b33926900447d5c1952a6e02187f33bdb318dd1/verif/regress/install-cva6.sh#L1) script is still needed.

cathales, 2023-12-19:
My 2 cents: when I refactored the submodule thing, I had tried to *just remove* the file but it didn’t work as it seems to export environment variables that we need. I did not investigate much.

MikeOpenHWGroup, 2023-12-20:
Yes, you are right @cathales, the file as it currently exists cannot be removed.  However, since it does not actually install the CVA6, it should at least be re-named, perhaps to `install-tools.sh` since its main function is to install Verilator and Spike.

My preference would be to move the functionality currently in `install-cva6.sh` to other scripts.

valentinThomazic, 2023-12-27:
`install-cva6.sh` (and its subscripts) and `install-riscv-dv.sh` set all environment variables required to run simulations with `cva6.py` and install the tools if they are not found. The only environment variable the user has to manually set is the toolchain location which is somehow convenient.
The fact they do both could indeed be misleading since they are named `install-X.sh`.

I think that the checks done to correctly set the environment variables related to a certain tool and to know wether the tool should be installed or not would probably be the same, explaining why it is done in the same script.

Maybe they could be turned into a `setup-sim.sh` setting the simulation tools and the appropriate environment variables. This could also be a one-time thing which would create a .env file loaded by `cva.py` so the users would not have to source theses scripts each time they want to simulate the cva6.

JeanRochCoulon, 2024-01-03:
#1741 has been merged. @MikeOpenHWGroup I let you close the issue (if you agree of course).

JeanRochCoulon, 2024-01-05:
As the file has been removed, I close the issue.

closed 2024-01-05 by JeanRochCoulon