openhwfoundation/cva6 #1735: [BUG] : MVENDORID Reset value

AyoubJalali, 2023-12-22:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, I'm a little confuse about the reset value of MVENDORID CSR : 

1. RTL read 0x00000602
2. The [doc](https://github.com/openhwgroup/cva6/blob/master/docs/01_cva6_user/CV32A6_Control_Status_Registers.rst#:~:text=Instruction%20Retired%20counter-,mvendorid,-0xF11) has 0x0 as reset value.

for sure there's a BUG but where ? 
IS it the CVA6 or the CSR documentation

MarioOpenHWGroup, 2024-01-08:
@JeanRochCoulon probably the documentation is wrong?

ASintzoff, 2024-01-10:
The RTL is correct (see issue #715).

AyoubJalali, 2024-01-10:
@JeanRochCoulon but in https://github.com/openhwgroup/cva6/blob/master/docs/01_cva6_user/CV32A6_Control_Status_Registers.rst ?

JeanRochCoulon, 2024-01-10:
#1754 requests to remove the CSR chapter from USer Manual. This GitHub issue can be close. Thank @AyoubJalali for the warning.

commit dc634c61 (2024-01-10): doc: update MVENDORID CSR value (fix #1735) (#1753)
closed 2024-01-10 by JeanRochCoulon