openhwfoundation/cva6 #1737: [BUG] Mechanism in `install-cva6.sh` for skipping Verilator build is broken

bluebandit21, 2023-12-22:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The shell script in `cva6/verif/regress/install-cva6.sh` contains a way to allow the user running it to skip the installation of Verilator.

https://github.com/openhwgroup/cva6/blob/6e79e20cc64451c5414a742aa6d697ddcb465b56/verif/regress/install-cva6.sh#L42-L46

https://github.com/openhwgroup/cva6/blob/6e79e20cc64451c5414a742aa6d697ddcb465b56/verif/regress/install-cva6.sh#L61-L63

Ostensibly, by reading that comment, calling install-cva6.sh as follows should work:
`VERILATOR_INSTALL_DIR=NO verif/regress/install-cva6.sh`

**However, this is not the case!** Later in the file is the following line, used to install Spike
https://github.com/openhwgroup/cva6/blob/6e79e20cc64451c5414a742aa6d697ddcb465b56/verif/regress/install-cva6.sh#L82-L82

Going into the referenced script, eventually
https://github.com/openhwgroup/cva6/blob/6e79e20cc64451c5414a742aa6d697ddcb465b56/verif/regress/install-spike.sh#L61-L61
calls `make` for a specific Makefile (as long as `SPIKE_INSTALL_DIR` is not "NO")

Inside that Makefile, 
https://github.com/openhwgroup/core-v-verif/blob/18c9d28cb96fbec8afb5e1bc5f344e9deb4a2cc8/vendor/riscv/riscv-isa-sim/Makefile.in#L72-L72
```Makefile
-I$(VERILATOR_INSTALL_DIR)/share/verilator/include/vltstd/
```

At this point, VERILATOR_INSTALL_DIR is "NO", so the include flag is set to (literally) `-INO/share/verilator/include/vlstd`
Obviously, this makes the build fail :)

-------------------------
Potential fixes I can see:
1) Make it clear that if VERILATOR_INSTALL_DIR is set to "NO", SPIKE_INSTALL_DIR must be as well.
2) Just use a different bash variable in `if [ "$VERILATOR_INSTALL_DIR" != "NO" ]; then `
e.g. SKIP_VERILATOR_INSTALL or something like that 🙂 

I'd make a PR directly doing one of the above, but figured I'd open an Issue instead since I'm not really familiar with how this project operates development-wise :)

bluebandit21, 2023-12-22:
For a little context as to how I even ran into this problem in the first place -- I had a heck of a time trying to install CVA6, in large part because I was doing so on a university server on which I did not have root access.

Consequently, I ended up re-running install-cva6.sh many many times, with the classic loop of "Run the script; it fails on line X; fix something; now it fails on line Y slightly further"

In the process, I noticed that Verilator was being re-built every time, eating up a large chunk of time before I got to the section actually failing.

I spotted that mechanism in install-cva6.sh which seemed useful for letting me bypass redoing the step already working, but as described in this issue that mechanism seems incompatible with the rest of the script :)

JeanRochCoulon, 2023-12-27:
related to #1727 

@valentinThomazic, the need of install-cva6.sh script is under question. Of course this script has not been done to be able to workaround the root access on a server. Maybe you have more visibility on it.

valentinThomazic, 2023-12-27:
@bluebandit21 it would be interesting to know for what you needed root access. 
Except for the prerequisites packages needed to build the RISCV toolchain, Spike and Verilator which have to be installed, I don't think that root access is needed to set the simulation environment.

bluebandit21, 2023-12-27:
Nothing interesting, literally just the prerequisite packages if I remember correctly :)

valentinThomazic, 2023-12-27:
Ok ! Also, Verilator should not be installed each time `install-cva6.sh` is run, are you sure this was the case ?

bluebandit21, 2023-12-27:
To be perfectly honest, I'm not lol

I made sure that everything in this issue was accurate and cited, but I didn't keep nearly as detailed notes while actually in the environment which led to me noticing this particular minor bug (the skip mechanism being fundamentally flawed)

I do see that there's a mechanism to prevent verilator from being re-built if it's already in the expected install location.

I'd have to do more testing to find out, certainly more than happy to do so if it would be helpful, though!

JeanRochCoulon, 2024-01-03:
For information #1741 has been merged. @bluebandit21 If you think this fixes the issue, please close the issue.

bluebandit21, 2024-01-03:
LGTM 👍 

As the relevant code no longer exists, closing this issue as resolved.
TY all for looking at this :)

closed 2024-01-03 by bluebandit21